cocotb/cocotb #3440: Loop unrolling too much.

firemountain154B, 2023-10-12:
Hello, 
When I want to test the module with loops more than 1024, cocotb returns an error, 
and told me I need to set --unroll-count above 1024, but I cannot find any docs, and I don't know where to set this counter.
[image]

marlonjames, 2023-10-12:
This looks like a simulator issue. There are ways to pass additional arguments to the simulator.
Could you please provide the requested info in the issue template?
cocotb version, os, simulator and version, python version, etc.

firemountain154B, 2023-10-12:
Thanks for reply, 
The cocotb version is 1.8.0
python version 3.10.7
OS is Linux
and simulator I use is verilator, 
I've just passed it to cocotb, not really sure about the version.
```
def main():
    sim = os.getenv("SIM", "verilator")
    verilog_sources = [...]
    runner = get_runner(sim)
    runner.build(
        verilog_sources=verilog_sources,
        hdl_toplevel="fixed_pvt",
        build_args=extra_args,
    )
    runner.test(
        hdl_toplevel="fixed_pvt",
        test_module="fixed_pvt_tb",
    )
```
How can I pass the arguments to verilator?

marlonjames, 2023-10-12:
Try adding ` --unroll-count` argument with a value higher than your loop counter to your `extra_args` list.

closed 2023-10-12 by cmarqu