cocotb/cocotb #3441: First and Combine can leak Tasks when the waiting Task is cancelled

marlonjames, 2023-10-13:
cocotb: master 5a892705222c651a84888a316eedd8f9da03ef6e

---

Because Waitables `First` and `Combine` are not actually triggers, and they rely on an `_InternalEvent` in the scheduler, they can leak Tasks.

If a `Task` that is awaiting a `Waitable` is cancelled, only the `_InternalEvent` is unprimed, and any waiter Tasks continue until the `Trigger` they are waiting on fires, or the test ends and they are cleaned up.

Consider this test:
```py3
@cocotb.test()
async def test_cancel_task_waiting_on_first(_):
    """Test that cancelling a task waiting on First trigger correctly unprimes the trigger and cleans up waiter tasks"""

    async def waiter():
        t1 = Timer(1, 'us')
        t2 = Timer(2, 'us')
        await First(t1, t2)

    task = cocotb.start_soon(waiter())

    await Timer(100, 'ns')
    task.cancel()

    assert len(cocotb.scheduler._trigger2coros) == 2

    await Timer(10, 'us')

    assert len(cocotb.scheduler._trigger2coros) == 0
```
In this example, not only does cancelling `task` leak 2 waiter Tasks, but they **both** fire by calling `_InternalEvent.set()`, since `task` is no longer around to cancel the remaining waiter task after the first one fires.

The impact is fairly low, since no `Task` is waiting on the `_InternalEvent` when it fires.

ktbarrett, 2023-10-14:
Could this be solved by adding a `deprime` overload to the `AggregateWaitable` class?

marlonjames, 2023-10-14:
The problem is that the scheduler only has the `_InternalEvent` trigger (because Waitables are not triggers at all), and that `_InternalEvent` is local to the coroutine `_wait()` that is running in the `Task`.

I see two options:
1. Keep a list on the `AggregateWaitable` object of the tasks that are created in `_wait()`, and create `_InternalEvent.unprime()` that calls something on the parent `AggregateWaitable` to cancel those tasks.
2. Implement throwing `CancelledError` into the Task and do the clean up in `_wait()`.

ktbarrett, 2023-10-14:
I tend to prefer expressing behavior structurally in code (`CancelledError`) rather than methods.

ktbarrett, 2023-11-14:
Is there a particular reason we don't treat Waitables as Triggers though? I feel like killing the waiter tasks would be very appropriate in a deprime method.

marlonjames, 2023-11-16:
I think they should be triggers so that they can be `unprime`d and be able to properly clean up.

PR #4542 (2025-03-19): Fix Combine and First leaking Tasks if cancelled
Closes #3441.

If a Task awaiting a Combine or First is cancelled, the waiter tasks weren't cleaned up. Now they are. Also fixed a small bug in First where if a waiter task was cancelled it would continue executing the `on_done` callback causing another exception to bubble up through the scheduler.
closed 2025-03-19 by ktbarrett