cocotb/cocotb #3445: Question: How to pass python options to python

Febbe, 2023-10-17:
While debugging, I get a warning, that
```
0.00s - Debugger warning: It seems that frozen modules are being used, which may
0.00s - make the debugger miss breakpoints. Please pass -Xfrozen_modules=off
0.00s - to python to disable frozen modules.
0.00s - Note: Debugging will proceed. Set PYDEVD_DISABLE_FILE_VALIDATION=1 to disable this validation
```
How can I pass `-Xfrozen_modules=off` to python. I think this could be the issue I have, that some coroutines aren't resumed, after their resume condition is true.

cmarqu, 2023-10-29:
Unfortunately, it seems that this currently cannot be set by an environment variable (https://docs.python.org/3/using/cmdline.html#environment-variables, https://github.com/python/cpython/issues/111374).

I'm not sure if there is anything that can be done until they fix that since we are using `libpython` for simulation, not the `python` binary.

marlonjames, 2023-11-30:
`Py_InitializeFromConfig` is 3.8+

ktbarrett, 2024-06-18:
xref https://github.com/cocotb/cocotb/issues/3643#issuecomment-1879758728.