cocotb/cocotb #3446: Multidimensional unpacked port

scorbetta, 2023-10-18:
I'm running cocotb version 1.7.2 on a 64-bit Ubuntu 22.04 machine w/ Python 3.10.12. I'm using Verilator version 5.014 to run the simulations.

My DUT has the following pinout:
```
module LAYER #(
    parameter NUM_INPUTS = 1,
    parameter NUM_OUTPUTS = 1,
    parameter WIDTH = 8
)
(
    input CLK,
    input RSTN,
    input signed [WIDTH-1:0] VALUES_IN [NUM_INPUTS],
    input signed [WIDTH-1:0] WEIGHTS_IN [NUM_OUTPUTS] [NUM_INPUTS],
    input VALID_IN
);
```

and the `test_layer.py` script contains the following snippet:

```
await RisingEdge(dut.CLK)
for idx in range(num_inputs):
    dut.VALUES_IN[idx].value = int(random_values_in_str[idx], 2)

for odx in range(num_outputs):
    for idx in range(num_inputs):
        dut.WEIGHTS_IN[odx][idx].value = int(random_weights_in_str[odx][idx], 2)
```

where `random_values_in_str` and `random_weights_in_str` are vector respectively matrix generated earlier.

The values on `VALUES_IN` go right throught the DUT, while the line assigning `WEIGHTS_IN` gives the following error:

```
File "/home/user/.local/lib/python3.10/site-packages/cocotb/handle.py", line 370, in __getattr__
    raise AttributeError(f"{self._name} contains no object named {name}")
AttributeError: LAYER contains no object named WEIGHTS_IN
```

By removing the assignment to `WEIGHTS_IN`, the simulation runs as expected, and from the VCD file I can see the `WEIGHTS_IN` matrix is actually there. The above works if the two dimensions on the unpacked side of the `WEIGHTS_IN` port are modified to a one-dimension:

```
input signed [WIDTH-1:0] WEIGHTS_IN [NUM_OUTPUTS*NUM_INPUTS],
```

What am I missing?

marlonjames, 2023-10-18:
Verilator does not support multi-dimensional unpacked arrays in its VPI implementation.

We need an upstream issue for this.

ktbarrett, 2023-10-20:
Right. And this is not the first issue about this that has been reported. We should consolidate.

jamesrbailey, 2025-01-14:
Verilator was just updated with "_Support multidimensional array access via VPI_" https://github.com/verilator/verilator/pull/5573

In principle this may improve array support with cocotb. However, with the following unpacked-unpacked-unpacked-packed testcase:

```verilog
module dut (
    input  logic [12:0] i_test[2][2][2],
    output logic [12:0] o_test[2][2][2]
);
    timeunit 1ns; timeprecision 1ps;

    assign o_test = i_test;

endmodule
```

```python
import cocotb
from cocotb.triggers import Timer

@cocotb.test()
async def mda_test(dut):
    dut._log.info(f"len(i_test) = {len(dut.i_test)}")
    dut._log.info(f"len(i_test[0]) = {len(dut.i_test[0])}")
    dut._log.info(f"len(i_test[0][0]) = {len(dut.i_test[0][0])}")
    dut._log.info(f"len(i_test[0][0][0]) = {len(dut.i_test[0][0][0])}")

    values = [[[1,2]]*2]*2
    dut._log.info(f"Attempting to drive {values}")

    dut.i_test.value = values

    await Timer(0.5, units="ns")
```

produces the following failure:
```
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:108  in set_program_name_in_venv        Using Python virtual environment interpreter at /home/jbailey/.cache/pypoetry/virtualenvs/nc-lf-lYkeGZgO-py3.12/bin/python
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on Verilator version 5.033 devel
     0.00ns INFO     cocotb                             Running tests with cocotb v1.9.1 from /home/jbailey/.cache/pypoetry/virtualenvs/nc-lf-lYkeGZgO-py3.12/lib64/python3.12/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1736869848
     0.00ns INFO     cocotb.regression                  Found test sanity.mda_test
     0.00ns INFO     cocotb.regression                  running mda_test (1/1)
     0.00ns INFO     cocotb.dut                         len(i_test) = 2
     0.00ns INFO     cocotb.dut                         len(i_test[0]) = 2
     0.00ns ERROR    gpi                                Unable to get range for indexable object
     0.00ns INFO     cocotb.dut                         len(i_test[0][0]) = 0
     0.00ns INFO     cocotb.dut                         len(i_test[0][0][0]) = 13
     0.00ns INFO     cocotb.dut                         Attempting to drive [[[1, 2], [1, 2]], [[1, 2], [1, 2]]]
     0.00ns INFO     cocotb.regression                  mda_test failed
                                                        Traceback (most recent call last):
                                                          File "/home/jbailey/workspaces/cocotb-mda/build/aw_dsp-nc-lf_sanity_1.0.0/default/sanity.py", line 20, in mda_test
                                                            dut.i_test.value = values
                                                            ^^^^^^^^^^^^^^^^
                                                          File "/home/jbailey/.cache/pypoetry/virtualenvs/nc-lf-lYkeGZgO-py3.12/lib64/python3.12/site-packages/cocotb/handle.py", line 201, in __setattr__
                                                            return object.__setattr__(self, name, value)
                                                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                                                          File "/home/jbailey/.cache/pypoetry/virtualenvs/nc-lf-lYkeGZgO-py3.12/lib64/python3.12/site-packages/cocotb/handle.py", line 489, in value
                                                            self._set_value(value, cocotb.scheduler._schedule_write)
                                                          File "/home/jbailey/.cache/pypoetry/virtualenvs/nc-lf-lYkeGZgO-py3.12/lib64/python3.12/site-packages/cocotb/handle.py", line 706, in _set_value
                                                            self[self_idx]._set_value(value[val_idx], call_sim)
                                                          File "/home/jbailey/.cache/pypoetry/virtualenvs/nc-lf-lYkeGZgO-py3.12/lib64/python3.12/site-packages/cocotb/handle.py", line 706, in _set_value
                                                            self[self_idx]._set_value(value[val_idx], call_sim)
                                                          File "/home/jbailey/.cache/pypoetry/virtualenvs/nc-lf-lYkeGZgO-py3.12/lib64/python3.12/site-packages/cocotb/handle.py", line 699, in _set_value
                                                            raise ValueError(
                                                        ValueError: Assigning list of length 2 to object unknown of length 0
     0.00ns INFO     cocotb.regression                  **************************************************************************************
                                                        ** TEST                          STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
                                                        **************************************************************************************
                                                        ** sanity.mda_test                FAIL           0.00           0.00          1.65  **
                                                        **************************************************************************************
                                                        ** TESTS=1 PASS=0 FAIL=1 SKIP=0                  0.00           0.13          0.01  **
                                                        **************************************************************************************
```

It seems that cocotb/verilator is able to determine the first two unpacked dimensions length correctly, but the third dimension's length cannot be determined. I have not dug into this further yet. It seems like an improvement, but still not functional at least for this particular case.

jamesrbailey, 2025-01-14:
It seems yes: https://github.com/verilator/verilator/blob/master/test_regress/t/t_vpi_multidim.v

I will try to spend some time root-causing, as I am very excited to have this working.

ksstrike, 2025-01-15:
Hello, author of the PR in question here, I'd be glad to help sort this issue out.
The rootcause here seems to be how vpi_iterate(vpiRange) is handled, i.e. for `i_test` Verilator will return `([0:1], [0:1], [0:1])` (in line with what's expected by cocotb), but for `i_test[0]` it returns `([0:1], [0:1])`, iterating **only** over the ranges that remain after indexing. Cocotb seems to expect all of the ranges to get returned irrespective of the current indexing depth: https://github.com/cocotb/cocotb/blob/stable/1.9/cocotb/share/lib/vpi/VpiCbHdl.cpp#L161

The SystemVerilog standard looks a bit ambiguous (see IEEE Std 1800-2017, 37.17), so I tried to see what Questa and Icarus have to say on the matter -- unfortunately, neither of them can really do "partial" indexing for unpacked arrays, which is what we're dealing with here.

It'd be pretty easy to change Verilator's behavior to match what cocotb is doing, but I'd like to have some reference, e.g. which simulators work properly with cocotb's vpi_iterate(vpiRange) handling -- or at least a solid argument for why my interpretation of the standard is flawed.

ktbarrett, 2025-01-15:
@ksstrike This is probably done because some simulators do not return handles for the intermediate array objects. We have things known as pseudo-regions to handle this. The handle these pseudo handles use is always the top-level array object handle.

If Verilator is actually returning intermediate objects that's good actually. Since the logic exists for cocotb-specific simulator workarounds I think the correct answer is to introduce more workarounds in cocotb.

jamesrbailey, 2025-01-15:
I can confirm that hacking cocotb to use the full non-array-dimension-scoped hdl_name does seem to make my testcase function.

https://github.com/jamesrbailey/cocotb/commit/b1d4fb4df42754ac33c89f170a6663a65901e15f#diff-a50fc3392c403c50e782febed83d1bcc6a5f7bd8caa99aae42be5939dfa0ddd0L61

Not sure if this is the correct workaround but it's a good sign.

ktbarrett, 2025-01-16:
@jamesrbailey Be careful with that change. It's going to return the wrong thing if you have multiple arrays, e.g. `top/gen[0]/array[0][1]`.

ktbarrett, 2025-04-14:
This may be fixed by #4548. If not, open a new issue.

closed 2025-04-14 by ktbarrett