cocotb/cocotb #3453: Internal error when top-module is called TOP

scorbetta, 2023-10-24:
I am compiling a SystemVerilog design using Verilator version `Verilator 5.014 2023-08-06 rev v5.014` on Ubuntu 22.04.

My design is made of few modules, with a top-level wrapper named `TOP` that instantiates the whole design:

```
// TOP.sv
`timescale 1ns/100ps

module TOP
(
    input                 CLK,
    input                 RSTN,
    input [23:0]      DATA_IN,
    input                 VALID_IN,
    output [23:0]   DATA_OUT,
    output               VALID_OUT
);

    MY_DESIGN MY_DESIGN_0 (
        .CLK               (CLK),
        .RSTN             (RSTN),
        .DATA_IN       (DATA_IN),
        .VALID_IN      (VALID_IN),
        .DATA_OUT   (DATA_OUT),
        .VALID_OUT  (VALID_OUT)
    );
endmodule

```

The following command fails:

```
verilator -cc --exe -Mdir sim_build --top-module TOP --public-flat-rw --prefix Vtop -o Vtop -I./src +1800-2017ext+sv --autoflush -Wno-WIDTHEXPAND --timescale 1ns/1ps [my_sv_files_here] src/TOP.sv
```

with the following output:

```
%Error: Internal Error: src/LOG.sv:37:13: ../V3LinkDot.cpp:973: Can't find instance insertion point at 'MY_DESIGN_0' in: 'MY_DESIGN_0.FIR_TIER.genblk1[0].FIR'
   37 |             FIR (
      |             ^~~~~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

However, the same does not happen if the module named within the `TOP.sv` file is changed to, e.g., `TOP_0`.

ktbarrett, 2023-10-24:
This sounds like an issue with Verilator, not cocotb.

marlonjames, 2023-10-24:
Try changing your top-level module name to something other than `TOP`.

Because cocotb uses `--public-flat-rw` to get VPI access to the top-level ports, Verilator exposes the special `TOP` module wrapper. There may be a problem with both having the same name.

closed 2023-10-24 by ktbarrett