cocotb/cocotb #3462: wave.do file, runner.test

thenameisT, 2023-10-30:
When running test using runner.test in GUI mode, is there a way to pass wave.do file and run -all ?
Whenever I run using GUI mode, I have to manually type _do wave.do_ and then _run -all_(in questa), I could not find a way to pass it to runner.test as do_script. And I also tried cocotb-test. 

I am using 
cocotb 1.8.0
python 3.11.4
Questa sim
Conda env

closed 2023-11-28 by thenameisT

klemen-erjavec-bl, 2024-02-08:
Did you found a solution for this? I was trying to pass the script to runners test_args, but apparently that command 'adds '\' before special characters and spaces', i think [as_tcl_value](https://docs.cocotb.org/en/v1.8.1/_modules/cocotb/runner.html) is called.

So I could not pass 'do ./wave.do' as test_args because this gets converted to  '-do\ ./wave.do'.

thenameisT, 2024-02-08:
Yes, I had to make changes to runner.py, check out the link
https://github.com/cocotb/cocotb/pull/3344

Cepadues, 2024-10-18:
Another solution would be to split the  'do ./wave.do' command into 2 strings : "do" "./wave.do" and pass them to the runner.