cocotb/cocotb #3467: Riviera-PRO: cbEndOfSimulation callbacks don't (always) come from main thread

imphil, 2023-10-31:
### Symptom
The test `test_custom_entry`, which tests that custom GPI entry points work, fails when using Riviera-PRO: it hangs at shutdown, never going beyond the following messages printed on stdout:

```
# KERNEL: sample_module: mybit has been updated, new value is 1
# KERNEL: sample_module: mybits has been updated, new value is 11
# KERNEL: Simulation has finished. There are no more test vectors to simulate.
endsim
```

### Analysis (symptoms)
GDB shows the hang to be in the threading module:

```
(gdb) py-bt
Traceback (most recent call first):
  File "/usr/lib64/python3.11/threading.py", line 1583, in _shutdown
    lock.acquire()
```

Reducing the test case shows that the test hangs as soon as the `logging` module is imported, which itself employs some locks.

### Analysis (root cause)

Riviera-PRO (all versions that I tested, up to the latest 2023.10) calls to the `cbEndOfSimulation` from a different thread than the other VPI callbacks. The similar VHPI callback shows the same behavior in my testing.

* `gpi_entry_point()`, a VPI entry point function (registered in `vlog_startup_routine`) is called from the "Main simulator" thread.
* `gpi_embed_end()`, a function registered with VPI as end-of-simulation callback (`cbEndOfSimulation`) is called from a "riviera_simulat" thread.

VPI (and VHPI) are single-threaded APIs: all callbacks should come from the same simulator thread. That's something cocotb is relying on here, and that's why the test fails.

### Impact

This issue only affects users with a custom GPI entry point, which does only a subset of work we're doing in cocotb's regular entry point. As soon as we use the "regular" GPI entry point with the scheduler and all of the "real" machinery, this problem disappears. We're likely ending up in another shutdown path in Riviera in these cases.

marlonjames, 2023-11-01:
This looks related to https://github.com/cocotb/cocotb/issues/1859

imphil, 2023-11-01:
Filed SPT82314 with Aldec.

imphil, 2024-06-11:
Fixed in Riviera-PRO 2024.04 and confirmed on my machine. Closing.

closed 2024-06-11 by imphil