cocotb/cocotb #3469: problem of Makefile.vcs

curliph, 2023-11-01:
Inside of Makefile.vcs,  now is:
```
# Compilation phase
$(SIM_BUILD)/simv: $(VERILOG_SOURCES) $(SIM_BUILD)/pli.tab $(CUSTOM_COMPILE_DEPS) | $(SIM_BUILD)
	cd $(SIM_BUILD) && \
	TOPLEVEL=$(TOPLEVEL) \
	$(CMD) -top $(TOPLEVEL) $(PLUSARGS) +acc+1 +vpi -P pli.tab +define+COCOTB_SIM=1 -sverilog \
	-timescale=$(COCOTB_HDL_TIMEUNIT)/$(COCOTB_HDL_TIMEPRECISION) \
	$(EXTRA_ARGS) -debug -load $(shell cocotb-config --lib-name-path vpi vcs) $(COMPILE_ARGS) $(VERILOG_SOURCES)
```
We do compile with 'cd $(SIM_BUILD)' ,  so  $(VERILOG_SOURCES) will not be seen there.
The updated maybe looks like follow:
```
# Compilation phase
$(SIM_BUILD)/simv: $(VERILOG_SOURCES) $(SIM_BUILD)/pli.tab $(CUSTOM_COMPILE_DEPS) | $(SIM_BUILD)
	TOPLEVEL=$(TOPLEVEL) \
	$(CMD) -top $(TOPLEVEL) $(PLUSARGS) +acc+1 +vpi -P **$(SIM_BUILD)/pli.tab** +define+COCOTB_SIM=1 -sverilog \
	-timescale=$(COCOTB_HDL_TIMEUNIT)/$(COCOTB_HDL_TIMEPRECISION) **-o $(SIM_BUILD)/simv** \
	$(EXTRA_ARGS) -debug -load $(shell cocotb-config --lib-name-path vpi vcs) $(COMPILE_ARGS) $(VERILOG_SOURCES)
```

That's we remove the 'cd $(SIM_BUILD)' and just redirect those output to right directories.

curliph, 2023-11-01:
Got, $(PWD) is a soultion.
there are projects (on github, e.g, cocotbext-axi) not following the $(PWD) rule
The above solution seems to work on those cases. (w/ or w/o $PWD).
Anyway, I will try to following $(PWD) rule. 
Thanks for your work.

closed 2023-11-01 by curliph

imphil, 2023-11-01:
You're welcome, enjoy using cocotb!