cocotb/cocotb #3483: long names for input and output port cause cocotb.handle simhandlebase recognition problem

ll550, 2023-11-05:
cocotb ver 1.8.1 stable
OS：win10 professional 64bit 22H2
simulator: modelsim 64bit 10.6d
python: 3.11.5 (anaconda 23.9.0)

Hi there,

I tried to modify the example test_matrix_multiplier.py to suit my own verilog design. However, I found the following lines may cause the recognition problem of  SimHandleBase, if the names of the input and output port of the verilog module is a bit longer with underscores.
For example,  

self.output_mon = DataValidMonitor(
            clk=self.dut.clk, valid=self.dut.valOut, datas=dict(G=self.dut.**F_col_updated**)
        )
If I change the F_col_updated to F_col, everything will be fine.

I am quite new to cocotb, and not very sure what is happening here.  But I guess it would be ok to report here, if someone else has a similar problem.

imphil, 2023-11-05:
Hi @ll550, thanks for trying out cocotb!

It feels unlikely that the length of the name is actually the problem here. Are you sure that you updated your Verilog code with the same name? Remember Verilog is case-sensitive! 

To help you further, please post the full code you're using, and the full error message you're getting when running the code.

ll550, 2023-11-06:
Hi @imphil,

I guess this issue could be closed. The problem I encoutered was caused by my code writting, once I completed the tester with _check() as did in the example. The errors have not shown any more. Sorry to bother.

closed 2023-11-06 by marlonjames