cocotb/cocotb #3492: docs: linkcheck broken links

marlonjames, 2023-11-07:
`(   release_notes: line  539) broken    https://apps.fedoraproject.org/packages/python-cocotb - 404 Client Error: Not Found for url: https://packages.fedoraproject.org//python-cocotb`
https://github.com/cocotb/cocotb/blob/3467ee5420bdef8a981bd6914bce30f258fe7131/docs/source/release_notes.rst?plain=1#L539
Looks like new link is https://packages.fedoraproject.org/pkgs/python-cocotb/

---

`(   release_notes: line  588) broken    https://lists.librecores.org/pipermail/cocotb/2019-January/000053.html - HTTPSConnectionPool(host='lists.librecores.org', port=443): Max retries exceeded with url: /pipermail/cocotb/2019-January/000053.html (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7fd8214c4d60>: Failed to establish a new connection: [Errno 113] No route to host'))`
https://github.com/cocotb/cocotb/blob/3467ee5420bdef8a981bd6914bce30f258fe7131/docs/source/release_notes.rst?plain=1#L588-L589
Anybody have a copy of the 1.1 release announcement?

imphil, 2023-11-07:
```
Subject: Cocotb 1.1 is released!
Date: Wed, 23 Jan 2019 23:09:09 +0000
From: Philipp Wagner <lists@philipp-wagner.com>

Hi cocotb users,

I'm very pleased to announce the release of cocotb 1.1! You can get the
source code for this release from the GitHub releases page at
https://github.com/potentialventures/cocotb/releases/tag/v1.1.

This release is the culmination of work done by 50 contributors over a
little less than four years. During that time we merged 242 pull
requests, resulting in 257 files changed, with 25,426 insertions and
6,289 deletions recorded by git -- growing the codebase by 19,137 lines!
The continued interest in cocotb is also visible on the issues front:
180 issues were closed since the last release and 124 new ones were
opened.

So what's behind all these numbers? A lot of refactorings, bug fixes,
and new features! I tried, but I couldn't compile a list of the most
significant features. So let's do it the other way around: How about you
scroll through the "shortlog" below and let us know what your favorite
change is?

You might ask: why did it take so long to get this release out? Cocotb
has been maintained by Stu and Chris since the beginning of (cocotb)
time. Cocotb as we know it today exists because of their vision of how
to produce a great hardware testing framework. They put a lot of time
into developing high-quality code and maintaining cocotb -- the
popularity of cocotb today is the best indicator of how well that worked
out. But as the popularity of cocotb grew, so did the maintenance
effort. In order to be able to grow further, a team of dedicated
volunteers agreed to share the maintainership role with Stu and Chris.
Based on a well-documented contribution process we were able to pick up
speed again, merging a large number pull requests and giving feedback on
many open issues.

Today, cocotb development is flourishing with many hard issues being
tackled, with documentation being improved, and new features being
added. One of the issues being worked on is also a more streamlined
release process, making it possible to release more frequently.

For the next release we already have an exciting headline feature to
announce: cocotb installable through pip! With cocotb being listed on
PyPi it should be even easier to use and update cocotb, and to create
projects that depend on it.

A big thank you goes out to all contributors, none of this would have
been possible without your hard work. For anybody visiting FOSDEM next
week, say hello and I'll get you a drink of your choice!

Philipp

PS: To stay informed about the latest developments in cocotb-land
subscribe to the mailing list at
https://lists.librecores.org/listinfo/cocotb!

The following people contributed code to this release
=====================================================

* Alexander Spirin
* Alexis Jeandet
* Andreas Galauner
* Ard Wiersma
* Ben Rosser
* Bruno Kremel
* Chris Higgs
* Christoph Zimmermann
* Colin Marquardt
* David Stanford
* Eric Wieser
* Fabien Marteau
* Gordon McGregor
* J. R. Petrus
* Jeremy Herbert
* Jeroen van Straten
* Jevin Sweval
* Joey
* John Peck
* Jonathan Morarity
* Joseph Glover
* Lance Eftink
* Luke Darnell
* Marco Eppenberger
* Marek Cieplucha
* Martin Hofherr
* Martin Zabel
* Matt Abdul-Rahim
* Mike Wild
* Mustafa Furkan Kaptan
* Nathan Kohagen
* Nevada Sanchez
* Nick Gasson
* Oliver A. Gubler
* Patrick Farley
* Philipp Wagner
* Sebastian
* Sigve Tjora
* Sjors Hettinga
* Stefan Gänsler
* Stefan Wallentowitz
* Steve Murphy
* Stuart Hodgson
* Tomasz Hemperek
* Tristan Gingold
* blorgon9000
* cospan
* elgorwi
* gideon
* potentialventures

Shortlog
========

The shortlog is too large to be appended to this mail. Find it at
https://gist.github.com/imphil/d0dbd642c96e80a2c68d5d907c4cce4c!
_______________________________________________
Discussion mailing list
[Discussion@lists.librecores.org](mailto:Discussion@lists.librecores.org)
https://lists.librecores.org/listinfo/discussion
```

ktbarrett, 2023-11-30:
Closed by #3521, but I simply deleted the text referencing the release notes ([see here](https://github.com/cocotb/cocotb/pull/3521/files#diff-eb36f1b6f2ee2691f609c07d32a8d977e4052a9e47d0ec94a241c6fc001a5d5cL589)). If this is important, perhaps we should upload and backlog all release notes to the website? I was never on the mailing list, so I don't have any of them 😆.

closed 2023-11-30 by ktbarrett