cocotb/cocotb #3500: Passing string define from a runner fails

Lockedloop, 2023-11-08:
I am running `cocotb-1.8.0`  and trying to create a string macro via runner.
E.g. I want to achieve `define MACRO_NAME "some-string"` using runners and Questa simulator.
Adding an entry in the defines dictionary `defines["MACRO_NAME"] = "some_string"` results in `"+define+MACRO_NAME=some_string"`.
I also tried `defines["MACRO_NAME"] = "\"some_string\""` , resulting in `"+define+MACRO_NAME=\"some_string""`, but the simulation fails with the error message
```
 (vlog-7300) Actual expression for parameter 'MACRO_NAME' in module instance 'my_module' must be constant.
```
Everything works  fine for integer parameters, so it seems to me it is an issue of the "stringification" in the Questa simulator function in `runner.py`.

Thank you!

ktbarrett, 2023-11-08:
Addressed by #3391.

Lockedloop, 2023-11-08:
I see, it seems to be the same problem. Closing this issue and keeping an eye out at #3391 for a solution. Thanks!

closed 2023-11-08 by Lockedloop