cocotb/cocotb #3504: Making it easier to simulate AMD-Xilinx IPs in cocotb using questa

mvsoliveira, 2023-11-10:
Dear developers,

Vivado has an excellent interface to questa so that one can switch a RTL simulation from xsim to questa with few clicks. Vivado automatically generates compile, elaborate, and simulate TCL scripts to take into account all files needed for each IP in questa. In some cases there are many files, see the Aurora implementation attached. 

[c2c_3125_exdes_tb_simulate.do.txt](https://github.com/cocotb/cocotb/files/13313919/c2c_3125_exdes_tb_simulate.do.txt)
[c2c_3125_exdes_tb_elaborate.do.txt](https://github.com/cocotb/cocotb/files/13313921/c2c_3125_exdes_tb_elaborate.do.txt)
[c2c_3125_exdes_tb_compile.do.txt](https://github.com/cocotb/cocotb/files/13313922/c2c_3125_exdes_tb_compile.do.txt)

It is well known that Xsim does not support the external interfaces required by cocotb, so cocotb can't support xsim as a simulator. On the other hand, using the FPGA-vendor tool as part of the process of configuring a cocotb simulation would be very helpful. This way all the complexity of defining every file needed for compiling several IPs would be automatically addressed in an FPGA vendor tool like VIvado. 

How do other people address this use case? I imagine would be a big effort to support officially something like this. However, based on the files I attached, would you have a suggestion on efficiently reusing these TCL files generated by VIvado in the cocotb environment or maybe easily translating to something compatible to cocotb Makefiles?

mvsoliveira, 2023-11-10:
Hi Kaleb,

It is super interesting what you just shared. I was not aware about this feature. 
Many thanks!!!

ktbarrett, 2023-11-30:
Feel free to reopen if you ran into any trouble.

mvsoliveira, 2024-01-01:
Hi @ktbarrett, talking about better ways of doing this, Vivado allows one to customize the `vsim` arguments, so the cocotb simulation can be started seamlessly from the Vivado GUI without modifying any TCL script following the instructions you shared.

ktbarrett, 2024-01-02:
Cool! I'll have to remember that.