cocotb/cocotb #3522: Can not found clocking block in cocotb

sordhlm, 2023-11-23:
cocotb version: 1.8.1
smulator: VCS
Python: 3.9.17

Hi, I am using cocotb to run some verification task right now. I intended to use interface in Verilog top.
It's ok in my python script to access signals in my interface, but it reported error when I tried to access clocking block.
Could anybody help check if cocotb can support this?
Here is what I do:
I created a interface like this:
```
interface dut_if (input clk);
    logic [15:0] dout;
    logic [15:0] din;
    logic ld, inc, rst_n;

    clocking cb1 @(posedge clk);
        default input #1ns output #1ns;
        output dout;
        input din;
        input ld, inc, rst_n;
    endclocking
endinterface
```
then I accessed this clocking block in my python script:
```
vif.cb1.din.value  = 1
```
Then it reported error:
```
File "/ic/software/tools/python3/3.9.17/lib/python3.9/site-packages/cocotb/handle.py", line 370, in __getattr__
    raise AttributeError(f"{self._name} contains no object named {name}")
    AttributeError: dut_if0 contains no object named cb
```

ktbarrett, 2023-11-23:
It's possible that clocking blocks are not supported by cocotb yet. We would need to know what VPI type is returned and if VCS even returns a handle to clocking blocks. Have you tried this with other simulators?

If you can, can you run a simple test which simply tries to access the signal while setting `COCOTB_LOG_LEVEL=DEBUG` and post the results here? That should allow us to see if the block is found and what its type is.

marlonjames, 2023-11-25:
```c
#define vpiClockingBlock                      65
```
It is a scope, so we should be able to support it.

ktbarrett, 2023-11-25:
I guess the real question is what are you intending to find under the clocking block? Supporting discovering what the inputs and outputs and the clocking info is probably not going to be feasible.

sordhlm, 2023-11-28:
Hi, here is the print info after I enbaled the debug info.
```
20.00ns WARNING  gpi                                VPI: Not able to map type vpiClockingBlock(650) to object.
```
It seems this is not supported yet.
I checked this for I wanted to add a little delay when I drived signals or got the value from signals;