cocotb/cocotb #3524: xcelium exits with internal error: "Routine com_vpiNoOp_chkRH() is an invalid call."

wimiko, 2023-11-23:
As the title suggest, xcelium exits with an internal error
This happens with cocotb 1.8.1 and the latest github version (32b8581). 
1.7.2 and older were working fine. 
Tested with cadence xcelium 23.03-s003 and 23.03-s005

I also ran the tests in tests/test_cases.  `test_vhdl_access` and 'test_vhdl_zerovector' fail, all other tests seem to be passing

```
test_vhdl_zerovector with DEBUG=1 gives the following output:
     0.00ns INFO     cocotb.regression                  running test_long_signal (1/4)
                                                          Write and read a normal signal (longer than 0).
ERROR:          VHPI       NOTOTO
         Can't get handle to object of type vhpiBaseType from a reference handle of type vhpiRootInstK.
        ./vhdl_zerovector.vhdl, 17: a vhpiRootInstK named :
ERROR:          VHPI       NOTOTO
         Can't get handle to object of type vhpiSubtype from a reference handle of type vhpiRootInstK.
        ./vhdl_zerovector.vhdl, 17: a vhpiRootInstK named :
INTERNAL ERROR: VHPI       NOFUNC
         Routine com_vpiNoOp_chkRH() is an invalid call.
--- ncdbg_exit() has been called
     0.00ns ERROR    gpi                                FATAL: We are calling up again

        PLIVERBOSE API stats
                Handle counts
                      VHPI handles:  5
TOOL:   xrun(64)        23.03-s003: Exiting on Nov 23, 2023 at 16:42:14 CET  (total: 00:02:04)
make[1]: *** [results.xml] Error 1
```

ktbarrett, 2023-11-23:
Could you run a `git blame` to find the offending commit?

rkumarkk010, 2024-02-14:
Hello, 

Can someone provide the solution for this issue? It still exists, I am trying similar as the original author and i recieve the same error. Simulator Xcelium, Design are VHDL, python verison 3.8, cocotb 1.8.1. Attached the error file[
[image]
](url) . If there is already a solution then how do i fix this ?

ktbarrett, 2024-02-14:
Have you tried with a more recent version of the simulator? There were a number of fixes done recently.

rkumarkk010, 2024-02-14:
Can you please tell me, in which version of the xcelium simulator the issue is resolved ?

ktbarrett, 2024-02-14:
I am not sure it was fixed, I'm not associated with Cadence. Try the most recent version, if it doesn't work we will consider this issue still relevant.

imphil, 2024-02-14:
This is pretty sure a duplicate of https://github.com/cocotb/cocotb/issues/1076.

ktbarrett, 2024-02-14:
You may need to use the master version of cocotb, alternatively, add this option to your commandline https://github.com/cocotb/cocotb/commit/88151f270f3eeda9144bae9ae39d0018887ae8f2.

rkumarkk010, 2024-02-15:
As you sugested, I redirected my makefile to point the master version COCOTB makefile path, 

Question-1: However I still get the issue for this as attached. Also I notice the GPI_EXTRA arg path points to the "usr/local/..." where I donot have the master version, is this a problem? unfortunately I cannot change that path i bieleve,
[image]

Question -2 : I am still using the Xcelium 22.09-s001 version, and my designs are in VHDL file inculding the top, I was reading the last issue comments, I saw a work around could be to use a verilog top? does this resolve the issue ? 

Question -3 : Also, as you mentioned, in the attached image, I should not be able to see this issue in the latest version without doing the work around
[image]

wimiko, 2024-02-15:
I can confirm that this is fixed in xcelium 23.09-s004. Triggers on vhdl signals are also working now.

This also works without the -NEW_VHPI_PROPAGATE_DELAY option. Not sure what the option is supposed to do as it is not documented . We're not using the cocotb makefiles though, and our toplevel is vhdl

rkumarkk010, 2024-02-15:
But the testbench is cocotb ?

> I can confirm that this is fixed in xcelium 23.09-s004. Triggers on vhdl signals are also working now.
> 
> This also works without the -NEW_VHPI_PROPAGATE_DELAY option. Not sure what the option is supposed to do as it is not documented . We're not using the cocotb makefiles though, and our toplevel is verilog, the rest of the design is mostly vhdl

wimiko, 2024-02-15:
Yes, the testbench is in cocotb

rkumarkk010, 2024-02-15:
Okay, will check and update. Thanks

closed 2024-02-15 by ktbarrett