cocotb/cocotb #3525: GHDL specify architecture to use of VHDL entity

forensicgarlic, 2023-11-24:
As far as I can tell, when simulating VHDL using GHDL using the latest version of CoCoTB, there's no way to specify an architecture of a dut to use from CoCoTB, either the python test, or the Makefile to drive the test. Is that correct? 

It looks like it may be possible to add the architecture via make variable to the Makefile.ghdl, but i wanted to double check this isn't easily solved some other way before i start messing with the Makefiles and then try to figure out how it would also work for the new pythonic test runs.

ktbarrett, 2023-11-24:
Is that a separate switch in GHDL? Usually simulators allow you to specify using the top. Something like `library.entity.arch`.

forensicgarlic, 2023-11-25:
Right, GHDL will allow you to specify the architecture as a separate value passed to the `ghdl -r <entity> [optional_architecture]`

The Makefile.ghdl just passes in $(TOP_LEVEL). Without adding anything I can experiment with $(EXTRA_ARGS) and determine if that's an effective method.

forensicgarlic, 2023-11-26:
adding an architecture like `behavioral` to `$(EXTRA_ARGS)` doesn't work, i get an error like 

```
/usr/bin/ghdl -m behavioral  --workdir=sim_build -Psim_build --work=work inc16
/usr/bin/ghdl-mcode:error: cannot open behavioral
/usr/bin/ghdl-mcode:error: importation has failed due to compilation error
```

forensicgarlic, 2023-11-26:
changing Makefile.ghdl like this and setting `$(ARCH)` works. 

```
diff --git a/cocotb/share/makefiles/simulators/Makefile.ghdl b/cocotb/share/makefiles/simulators/Makefile.ghdl
index d6b6e7b8..62163c90 100644
--- a/cocotb/share/makefiles/simulators/Makefile.ghdl
+++ b/cocotb/share/makefiles/simulators/Makefile.ghdl
@@ -70,13 +70,13 @@ analyse: $(VHDL_SOURCES) $(CUSTOM_COMPILE_DEPS) | $(SIM_BUILD)
        $(foreach SOURCES_VAR, $(filter VHDL_SOURCES_%, $(.VARIABLES)), \
                $(CMD) -i $(GHDL_ARGS) $(COMPILE_ARGS) --workdir=$(SIM_BUILD) --work=$(SOURCES_VAR:VHDL_SOURCES_%=%) $($(SOURCES_VAR)) && ) \
        $(CMD) -i $(GHDL_ARGS) $(COMPILE_ARGS) --workdir=$(SIM_BUILD) --work=$(RTL_LIBRARY) $(VHDL_SOURCES) && \
-       $(CMD) -m $(GHDL_ARGS) $(COMPILE_ARGS) --workdir=$(SIM_BUILD) -P$(SIM_BUILD) --work=$(RTL_LIBRARY) $(TOPLEVEL)
+       $(CMD) -m $(GHDL_ARGS) $(COMPILE_ARGS) --workdir=$(SIM_BUILD) -P$(SIM_BUILD) --work=$(RTL_LIBRARY) $(TOPLEVEL) $(ARCH)
 
 $(COCOTB_RESULTS_FILE): analyse $(CUSTOM_SIM_DEPS)
        $(RM) $(COCOTB_RESULTS_FILE)
 
        MODULE=$(MODULE) TESTCASE=$(TESTCASE) TOPLEVEL=$(TOPLEVEL) TOPLEVEL_LANG=$(TOPLEVEL_LANG) \
-       $(SIM_CMD_PREFIX) $(CMD) -r $(GHDL_ARGS) --workdir=$(SIM_BUILD) -P$(SIM_BUILD) --work=$(RTL_LIBRARY) $(TOPLEVEL) --vpi=$(shell cocotb-config --lib-name-path vpi ghdl) $(SIM_ARGS) $(PLUSARGS)
+       $(SIM_CMD_PREFIX) $(CMD) -r $(GHDL_ARGS) --workdir=$(SIM_BUILD) -P$(SIM_BUILD) --work=$(RTL_LIBRARY) $(TOPLEVEL) $(ARCH) --vpi=$(shell cocotb-config --lib-name-path vpi ghdl) $(SIM_ARGS) $(PLUSARGS)
 
        $(call check_for_results_file)
```

PR #3541 (2023-11-29): Add  to GHDL makefile to simulate different VHDL architectures from the top level
Closes #3525.

This will make it possible with the MAKE build flow to specify the VHDL architecture to use for a specific test. Without this, the architecture would default to the last one in a VHDL file and can not be specified for COCOTB driven tests.
closed 2023-11-29 by ktbarrett