cocotb/cocotb #3540: GHDL array types not handling

franhervas99, 2023-11-28:
> cocotb version: 1.8.1
> OS: 22.04.1-Ubuntu
> simulator: GHDL 4.0.0-dev (tarball) [Dunoon edition]
> python: 3.10.12

Hello everyone,

I'm trying to do a simple test to see some signals behavior. The problem is that the user array types can't be accessed.
The vhdl file in question is this one:
```vhdl

library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

use work.neural_network_pkg.all;

entity input_layer is
    generic(
        g_samples : natural := 9
    );
    port(
        i_input     : in window(0 to g_samples-1);
        o_output    : out vector_of_signed14(0 to g_samples-1)
    );
end input_layer;

architecture behavior of input_layer is

    constant GAIN1 : unsigned(13 downto 0) := "11101111000001";
    constant YMIN1 : signed(15 downto 0) := "1111000000000000";

    signal s_input      : window(0 to g_samples-1);
    signal s_output     : vector_of_signed14(0 to g_samples-1);

    signal s_mult       : vector_of_signed32(0 to g_samples-1);
    signal s_add        : vector_of_signed16(0 to g_samples-1);

begin

    generator: for i in 0 to g_samples-1 generate
        s_mult(i) <= resize(signed(('0' & s_input(i)) * GAIN1), 32);
        s_add(i) <= resize(s_mult(i)(26 downto 13),16) + YMIN1;
        s_output(i) <= s_add(i)(13 downto 0);
    end generate;

    s_input <= i_input;
    o_output <= s_output;

end behavior;
```
The array types are defined in neural_network.pkg.
When I run a simple test file for seeing the items in the vhdl file (this file):
```python
@cocotb.test()
async def multi_nn_test(dut):
	print(dir(dut))
```
I get the following output:
```bash
rm -f results.xml
make -f Makefile results.xml
make[1]: Entering directory '/home/francisco/IFIC/ghdl/multiple_neural_network/test'
/opt/ghdl/bin/ghdl -i -fsynopsys  --workdir=work --work=work /home/francisco/IFIC/ghdl/multiple_neural_network/test/../work/neural_network_pkg.vhd /home/francisco/IFIC/ghdl/multiple_neural_network/test/../work/trace_pkg.vhd &&  \
/opt/ghdl/bin/ghdl -i -fsynopsys  --workdir=work --work=work /home/francisco/IFIC/ghdl/multiple_neural_network/test/../tanh_mem.vhd /home/francisco/IFIC/ghdl/multiple_neural_network/test/../shift_register.vhd /home/francisco/IFIC/ghdl/multiple_neural_network/test/../input_layer.vhd /home/francisco/IFIC/ghdl/multiple_neural_network/test/../hidden_layer_1.vhd /home/francisco/IFIC/ghdl/multiple_neural_network/test/../hidden_layer_2.vhd /home/francisco/IFIC/ghdl/multiple_neural_network/test/../output_layer.vhd /home/francisco/IFIC/ghdl/multiple_neural_network/test/../neural_network.vhd /home/francisco/IFIC/ghdl/multiple_neural_network/test/../bram_dual_port.vhd /home/francisco/IFIC/ghdl/multiple_neural_network/test/../multi_nn_v4.vhd && \
/opt/ghdl/bin/ghdl -m -fsynopsys  --workdir=work -Pwork --work=work input_layer
rm -f results.xml
MODULE=test_cocotb TESTCASE= TOPLEVEL=input_layer TOPLEVEL_LANG=vhdl \
 /opt/ghdl/bin/ghdl -r -fsynopsys --workdir=work -Pwork --work=work input_layer --vpi=/home/francisco/.local/lib/python3.10/site-packages/cocotb/libs/libcocotbvpi_ghdl.so  
loading VPI module '/home/francisco/.local/lib/python3.10/site-packages/cocotb/libs/libcocotbvpi_ghdl.so'
[... 66 lines cut ...]
		$(PWD)/../work/trace_pkg.vhd
	# VHDL_SOURCES_work+=$(PWD)/../work/trace_pkg.vhd
	VHDL_SOURCES+=$(PWD)/../tanh_mem.vhd
	VHDL_SOURCES+=$(PWD)/../shift_register.vhd
	VHDL_SOURCES+=$(PWD)/../input_layer.vhd
	VHDL_SOURCES+=$(PWD)/../hidden_layer_1.vhd
	VHDL_SOURCES+=$(PWD)/../hidden_layer_2.vhd
	VHDL_SOURCES+=$(PWD)/../output_layer.vhd
	VHDL_SOURCES+=$(PWD)/../neural_network.vhd
	VHDL_SOURCES+=$(PWD)/../bram_dual_port.vhd
	VHDL_SOURCES+=$(PWD)/../multi_nn_v4.vhd
endif

EXTRA_ARGS = -fsynopsys

# TOPLEVEL is the name of the toplevel module in your Verilog or VHDL file
TOPLEVEL = input_layer

# MODULE is the basename of the Python test file
MODULE = test_cocotb

# include cocotb's make rules to take care of the simulator setup
include $(shell cocotb-config --makefiles)/Makefile.sim
```
Thank you in advance.

ktbarrett, 2023-11-28:
This might be related to #2594. The issue is that the VPI doesn't know how to create pseudo-regions for the generate block based on the fact that GHDL uses different syntax. Try running `dir` on the parent scope and then use `getattr(dut, "generator(0)")`. Alternatively, use try a different VHDL simulator like NVC, Questa, Riviera, Xcelium, etc.

> and also appears the message vpi_get_str: unhandled property so many times.

See #3426. That should be resolved before the next release since it's a simple enough fix.

I'm marking this as a dupe, but I'll keep this open until you find something that works.

ktbarrett, 2023-11-28:
You access `s_mult` *under* the generate block. So `getattr(dut, "generate(0)").s_mult`.

ktbarrett, 2023-11-28:
Ah, I see what you're saying. It seems that GHDL is optimizing those signals away. I don't know much about GHDL. Is there a flag you can add which increases the "access" or "visibility" of signals for debugging? For example, we have to add `+acc` to Questa runs so that internal signals like that aren't optimized away and show up in the debugging window and are available to cocotb.

franhervas99, 2023-11-29:
I didn't find nothing like that. For adding more information, I have generated the vpi log file. Is the next one:
```log
vpi_register_cb ({reason=cbStartOfSimulation, obj=0000000000000000}) = 0000560793DF97A0
vpi_register_cb ({reason=cbEndOfSimulation, obj=0000000000000000}) = 0000560793F12610
vpi call callback 0000560793DF97A0 cbStartOfSimulation
 vpi_get_vlog_info
 vpi_get_vlog_info
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 0ms
 vpi_get (vpiTimePrecision, 0000000000000000) = -15
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 0ms
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 0ms
 vpi_iterate (vpiModule, 0000000000000000) = 00005607943C9370
 vpi_scan (00005607943C9370) = 00005607943B45B0
 vpi_get (vpiType, 00005607943B45B0) = vpiModule
 vpi_get_str (vpiFullName, 00005607943B45B0) = "input_layer"
 vpi_free_object (00005607943C9370)
 vpi_get_str (vpiFullName, 00005607943B45B0) = "input_layer"
 vpi_get (vpiType, 00005607943B45B0) = vpiModule
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 0ms
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 0ms
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 0ms
 vpi_register_cb ({reason=cbAfterDelay, obj=0000000000000000, time=1ps}) = 00005607952CE570
vpi end callback 0000560793DF97A0
vpi call callback 00005607952CE570 cbAfterDelay
 vpi_free_object (00005607952CE570)
 vpi_get (vpiType, 00005607943B45B0) = vpiModule
 vpi_iterate (vpiNet, 00005607943B45B0) = 0000560794E75FD0
 vpi_get_str (vpiType, 00005607943B45B0) = "???"
 vpi_get_str (vpiFullName, 00005607943B45B0) = "input_layer"
 vpi_scan (0000560794E75FD0) = 0000000000000000
 vpi_iterate (vpiNetArray, 00005607943B45B0) = 0000000000000000
 vpi_iterate (48, 00005607943B45B0) = 0000000000000000
 vpi_iterate (116, 00005607943B45B0) = 0000000000000000
 vpi_iterate (29, 00005607943B45B0) = 0000000000000000
 vpi_iterate (25, 00005607943B45B0) = 0000000000000000
 vpi_iterate (47, 00005607943B45B0) = 0000000000000000
 vpi_iterate (526, 00005607943B45B0) = 0000000000000000
 vpi_iterate (618, 00005607943B45B0) = 0000000000000000
 vpi_iterate (683, 00005607943B45B0) = 0000000000000000
 vpi_iterate (100, 00005607943B45B0) = 0000000000000000
 vpi_iterate (34, 00005607943B45B0) = 0000000000000000
 vpi_iterate (129, 00005607943B45B0) = 0000000000000000
 vpi_iterate (vpiParameter, 00005607943B45B0) = 0000000000000000
 vpi_iterate (103, 00005607943B45B0) = 0000000000000000
 vpi_iterate (113, 00005607943B45B0) = 0000000000000000
 vpi_iterate (99, 00005607943B45B0) = 0000000000000000
 vpi_iterate (31, 00005607943B45B0) = 0000000000000000
 vpi_iterate (61, 00005607943B45B0) = 0000000000000000
 vpi_iterate (105, 00005607943B45B0) = 0000000000000000
 vpi_iterate (vpiPort, 00005607943B45B0) = 0000560794E75FD0
 vpi_scan (0000560794E75FD0) = 0000000000000000
 vpi_iterate (vpiInternalScope, 00005607943B45B0) = 0000560794E75FD0
 vpi_scan (0000560794E75FD0) = 000056079525CA40
 vpi_get_str (vpiName, 000056079525CA40) = "generator(0)"
 vpi_get (vpiType, 000056079525CA40) = vpiModule
 vpi_get (vpiType, 000056079525CA40) = vpiModule
 vpi_get_str (vpiName, 000056079525CA40) = "generator(0)"
 vpi_get_str (vpiDefName, 000056079525CA40) = null
 vpi_get_str (vpiDefFile, 000056079525CA40) = null
 vpi_get_str (vpiType, 000056079525CA40) = "???"
 vpi_scan (0000560794E75FD0) = 00005607947A7990
 vpi_get_str (vpiName, 00005607947A7990) = "generator(1)"
 vpi_get (vpiType, 00005607947A7990) = vpiModule
 vpi_get (vpiType, 00005607947A7990) = vpiModule
 vpi_get_str (vpiName, 00005607947A7990) = "generator(1)"
 vpi_get_str (vpiDefName, 00005607947A7990) = null
 vpi_get_str (vpiDefFile, 00005607947A7990) = null
 vpi_get_str (vpiType, 00005607947A7990) = "???"
 vpi_scan (0000560794E75FD0) = 000056079479B820
 vpi_get_str (vpiName, 000056079479B820) = "generator(2)"
[... 45 lines cut ...]
 vpi_get_str (vpiType, 0000560794B93830) = "???"
 vpi_scan (0000560794E75FD0) = 00005607949B7550
 vpi_get_str (vpiName, 00005607949B7550) = "generator(8)"
 vpi_get (vpiType, 00005607949B7550) = vpiModule
 vpi_get (vpiType, 00005607949B7550) = vpiModule
 vpi_get_str (vpiName, 00005607949B7550) = "generator(8)"
 vpi_get_str (vpiDefName, 00005607949B7550) = null
 vpi_get_str (vpiDefFile, 00005607949B7550) = null
 vpi_get_str (vpiType, 00005607949B7550) = "???"
 vpi_scan (0000560794E75FD0) = 0000000000000000
 vpi_register_cb ({reason=cbAfterDelay, obj=0000000000000000, time=1fs}) = 00005607952D7690
vpi end callback 00005607952CE570
vpi call callback 00005607952D7690 cbAfterDelay
 vpi_free_object (00005607952D7690)
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 1001fs
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 1001fs
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 1001fs
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 1001fs
 vpi_control (vpiFinish, 1)
vpi end callback 00005607952D7690
vpi call callback 0000560793F12610 cbEndOfSimulation
vpi end callback 0000560793F12610
```
As you can see the array signals are not appearing, neither the constants declared.
That could be a problem with the vpi interface?

franhervas99, 2023-11-29:
Okay! more information. If I give the option `SIM_ARGS=--trace-signals` to the simulator, I have the following output:
```log
...
.input_layer(behavior).s_add(7)(10) 000055BCB1F2FED0  e8 ------- last_event=0ms last_active=0ms val='X'; drv='X'
.input_layer(behavior).s_add(7)(9) 000055BCB1F2FF70  e8 ------- last_event=0ms last_active=0ms val='X'; drv='X'
.input_layer(behavior).s_add(7)(8) 000055BCB1F30010  e8 ------- last_event=0ms last_active=0ms val='X'; drv='X'
.input_layer(behavior).s_add(7)(7) 000055BCB1F300B0  e8 ------- last_event=0ms last_active=0ms val='X'; drv='X'
.input_layer(behavior).s_add(7)(6) 000055BCB1F30150  e8 ------- last_event=0ms last_active=0ms val='X'; drv='X'
.input_layer(behavior).s_add(7)(5) 000055BCB1F301F0  e8 ------- last_event=0ms last_active=0ms val='X'; drv='X'
.input_layer(behavior).s_add(7)(4) 000055BCB1F30290  e8 ------- last_event=0ms last_active=0ms val='X'; drv='X'
.input_layer(behavior).s_add(7)(3) 000055BCB1F30330  e8 ------- last_event=0ms last_active=0ms val='X'; drv='X'
.input_layer(behavior).s_add(7)(2) 000055BCB1F303D0  e8 ------- last_event=0ms last_active=0ms val='X'; drv='X'
.input_layer(behavior).s_add(7)(1) 000055BCB1F30470  e8 ------- last_event=0ms last_active=0ms val='X'; drv='X'
.input_layer(behavior).s_add(7)(0) 000055BCB1F30510  e8 ------- last_event=0ms last_active=0ms val='X'; drv='X'
.input_layer(behavior).s_add(8)(15) 000055BCB1F305B0  e8 ------- last_event=0ms last_active=0ms val='X'; drv='X'
.input_layer(behavior).s_add(8)(14) 000055BCB1F30650  e8 ------- last_event=0ms last_active=0ms val='X'; drv='X'
.input_layer(behavior).s_add(8)(13) 000055BCB1F306F0  e8 ------- last_event=0ms last_active=0ms val='X'; drv='X'
.input_layer(behavior).s_add(8)(12) 000055BCB1F30790  e8 ------- last_event=0ms last_active=0ms val='X'; drv='X'
.input_layer(behavior).s_add(8)(11) 000055BCB1F30830  e8 ------- last_event=0ms last_active=0ms val='X'; drv='X'
.input_layer(behavior).s_add(8)(10) 000055BCB1F308D0  e8 ------- last_event=0ms last_active=0ms val='X'; drv='X'
.input_layer(behavior).s_add(8)(9) 000055BCB1F30970  e8 ------- last_event=0ms last_active=0ms val='X'; drv='X'
...
```
It has a lot of information like that. I don't know if there could be the solution for the problem.

franhervas99, 2023-11-29:
OKAY!
I just realized how to access arrays. Basically, I was trying to access them by seeing on `dir(dut)`, but they were not appearing. If I do directly `print(dut.s_test.value)` (where s_test is an array-like type), I can see the full array. Also, if I do `print(dut.s_test.value[0])`, I can see the value at index 0.
Now I'm trying to modify those values and seeing what happens.
They also appear on the vpi log file when I try to access them and modify them.

The point is that the arrays don't appear if I try `dir(dut)`, but I have access to them if I do `dut.array.value`. There was my confusion.
Thank you!

closed 2023-11-30 by ktbarrett

ktbarrett, 2024-02-27:
We need a minimal failing test to reproduce the issue before we can do any sort of debugging on it. The original issue was resolved, please open a new issue.

alexfanqi, 2024-02-28:
I am having same issues as @franhervas99 reported one comment above.

minimal failing test
```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
package pkg_type is
type t_input_block is array (16 - 1 downto 0) of signed(8 - 1 downto 0);

end package pkg_type;

------------------------ 
library ieee;
use ieee.std_logic_1164.all;
entity lower is
	generic(
		WEIGHT : in std_logic_vector(8 - 1 downto 0)
	);
	port (
	s_i: in std_logic
			 );
end entity lower;

architecture rtl of lower is
begin
end architecture rtl;

------------------------ 
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
library work;
use work.pkg_type.all;
entity top is
	port (
	a_block_i: in t_input_block;
	e_i: in std_logic_vector (15 downto 0);
	c_block_o: out t_input_block
			 );
end entity top;

architecture rtl of top is
	signal b: t_input_block;
	signal d: std_logic_vector (7 downto 0);
begin
	b <= a_block_i;
	c_block_o <= b;

	generator: for i in 7 downto 0 generate
	begin
		d(i) <= '0';
		inst_lower: entity work.lower
		generic map(
			WEIGHT => std_logic_vector(to_unsigned(i, 8))
		)
		port map(
			s_i => d(0)
						);
	end generate generator;

end architecture rtl;
```
test bench
```
import cocotb

@cocotb.test()
async def test_top(dut):
    # only show signal d, generator and input e_i
    print(f"{dut._name}: \n{dir(dut)}")
    # show inst_lower
    print(f"generator: \n{dir(getattr(dut,'generator(0)'))}")
    # fail
    # print(f"generator: \n{dir(getattr(dut,'generator(0)').inst_lower)}")
    # ok
    print(f"inst_lower: \n{dir(getattr(getattr(dut,'generator(0)'),'inst_lower'))}")
    # ok
    getattr(getattr(dut,'generator(0)'),'inst_lower').s_i.value = 0

    # ok
    print(f"b: {dut.b[0].value}")
    print(f"e_i: {dut.e_i.value}")
    # fail
    print(f"c: {dut.c_block_o.value}")
    print(f"c: {dut.c_block_o[0].value}")
    print(f"a: {dut.a_block_i[0].value}")
```
makefile
```
export MODULE=test
export TOPLEVEL=top
all: top.vhd test.py
	ghdl -a -g top.vhd
	ghdl -e top
	ghdl -r top --vpi=`cocotb-config --lib-name-path vpi ghdl`
```