cocotb/cocotb #3544: Do we really need to wait until ReadWrite to apply writes if writes are inertial?

ktbarrett, 2023-11-29:
Title says it all. I think we could get rid of the ReadWrite stage writing of values and let inertial writes happen as they otherwise would. This would increase performance noticeably (1.5~2x I would think).

marlonjames, 2023-11-29:
I think this would cause subtle changes in behavior.
`Timer` is triggered early in the time step, before processes are run, and doing the writes at that time might change behavior, but I haven't looked at the event models in a while.

I wouldn't be against adding a setting that lets users switch to your proposed behavior.

ktbarrett, 2024-03-18:
> `Timer` is triggered early in the time step, before processes are run

So this would mean that writes applied by cocotb would be applied after the `Timer` callback and before the processes run, allowing them to be "seen" by processes? That would be less than ideal. IMO cocotb tasks should be treated as more or less parallel to processes in HDLs. 

But applying writes at the beginning of the `ReadWrite` stage means that cocotb tasks that operate directly after those writes in that `ReadWrite` stage aren't operating on the outcome of processes or cocotb tasks that applied writes in the previous "pseudo process execution" cycle.

PR #3875 (2024-06-08): Introduce mode to trust inertial writes
Closes #3544.

Just read the documentation. This provides better behavior and performance where it will work.

### TODO
- [x] fix tests for commercial sims
closed 2024-06-08 by ktbarrett