cocotb/cocotb #3551: Remove types from docstrings, use type annotations instead

cmarqu, 2023-12-02:
We should remove the types from the docstrings and use type annotations for Sphinx (may need to be configured).

Some of our docstring types are not quite correct, e.g. https://github.com/cocotb/cocotb/blob/b5eab5a0dc1b35497556c1bcc2c8c009feae805f/src/cocotb/clock.py#L59
(https://docs.cocotb.org/en/latest/library_reference.html#cocotb.clock.Clock)

Holarctic, 2023-12-21:
Hi @cmarqu,

I'm interested in contributing to cocotb and would like to work on issue #3551. I've reviewed the details and believe I can help. Could you guide me on how best to proceed?

cmarqu, 2023-12-21:
Hi @Holarctic, thanks for the offer!

I'd advise picking a file that has both type annotations in code and docstrings (stating the obvious) and then quickly opening a PR to get it merged, because the codebase is under a high rate of change right now.
Please make sure to keep the changes limited to exactly this task and with the minimum number of lines touched to keep the diff small (and the chance of merge conflicts low).

The `CONTRIBUTING.md` file is also interesting.

ktbarrett, 2023-12-22:
I will mention that #3578 touches a couple of these and I'm working on a PR for everything in handle.py that will include these changes.

commit b074e7f7 (2024-06-03): Remove types from docstrings

Where types were used in docstring, argument types are explicited
to help Sphinx when building documentation.

Fixes #3551