cocotb/cocotb #3562: Thoughts on adding git rev to dev version numbers?

toddstrader, 2023-12-06:
How would everyone feel about adding `git rev-parse --short HEAD` to `__version__`?  Currently it is `2.0.0.dev0` and I'm not sure what the final `0` is for, but I'm guessing it's to manually increment as one feels like?  What if instead it were `2.0.0.dev-d379318e`?  I'd propose that we build the file on the fly in `setup.py` but only add the git rev when the non-suffixed version (stored as a variable in `setup.py`) includes `dev`.

I can take a stab at this if people are onboard.

ktbarrett, 2023-12-07:
setuptools-scm did this. But it did a lot more than that, poorly. Maybe you can extract the logic it uses from it. Technically I think the correct version we should be using is `2.0.devN`, but I'd have to check the PEP. I'd like the N incremented for each commit as well.

marlonjames, 2023-12-07:
I would like this as well.

PR #3568 (2023-12-10): Add git rev to local version identifier
Adds the Git rev to the local version label.  There's a chicken-and-egg problem wrt PR numbers and creating news fragments.  I'll add that next.  Also, should I add a test for this?  I guess I could check that the `+` exists and there's something git rev-like if it's a `dev` version?

closes #3562
closed 2023-12-10 by imphil