cocotb/cocotb #3583: Separating symmetric and asymmetric tasks

ktbarrett, 2023-12-13:
## Background

Follow on to #2664.

As stated the original issue is that if a Task without a registered Join trigger throws an exception, the test is ended with the exception as the test exception. But if the Task has a Join trigger and an exception is thrown, that exception is propagated to the Task waiting on the Join. This is inconsistent and non-deterministic. One can intend to wait on the failing Task, but the Task can fail before the wait occurs, making it impossible to catch.

An example of the above gotcha:
```python
@cocotb.test()
async def test_fails(dut):
    async def fails_after_random_time():
        await Timer(randint(1, 100), 'ns')
        raise ValueError("pretend this was expected for valid reasons")
    task = cocotb.start_soon(fails_after_random_time())
    await Timer(50, 'ns')
    try:
        await task
    except ValueError:
        pass  # pretend we handle this
```

In the example, the `fails_after_random_time` task might throw before or after the task is await. If it throws before, the test fails. If it throws after, the exception is caught and handled. Of course, this example is contrived, but shows the inconsistent behavior.

## Currently slated solution

The current solution, from #2664, is that tasks created with `cocotb.start_soon` will *not* fail the test if they get an exception and will always store the exception for propagation. This follows how asyncio is implemented.

The issue with this approach is that there is no way currently to get the existing behavior of ending the test if the task throws an exception, which is commonly needed for verification components, such as Drivers, Monitors, Scoreboards, etc.

## Possible solution

One solution is to have two types of tasks: symmetric and asymmetric (terminology from Revisiting Coroutines, De Moura and Ierusalimschy). Symmetric tasks run at the same hierarchical level as the test and other tasks are not intended to directly wait on them. Asymmetric tasks are children of other tasks and are intended to be waited upon and have return info and exceptions propagated. In this setup, symmetric tasks getting an exception would fail the test, and asymmetric tasks getting an exception would capture the exception for propagation.

### Introducing `TaskGroup`s as asymmetric control, leave `cocotb.start_soon` as symmetric

This would do the opposite of what the `FutureWarning` in #2664 introduces: exceptions would *always* fail the test. There would need to be another interface for creating asymmetric tasks. Since we are considering adding `TaskGroup`s, which are similar to SystemVerilog's fork/join blocks, that would provide an excellent way of enforcing asymmetric Tasks.

For an example:
```python
@cocotb.test
async def test_example(dut):
    # if Clock's start task throws an exception, the test should fail
    # the driver isn't working as intended so the test can't continue
    clock_task = cocotb.start_soon(Clock().start())

    async with TaskGroup() as tg:

        tg.start_soon(some_coro())

        @tg.start_soon
        async def do_stuff():
            await Timer(10, 'ns')
            raise ValueError

        # waits for all child tasks to finish

    # since "do_stuff" throws ValueError, it is reraised here
    # ends the test_example task with the ValueError
```

Notes:
* If a TaskGroup were waiting when the Task containing that TaskGroup were cancelled, a CancelledError would be throw into the TaskGroup __aexit__, which would then cancel all the children in turn.
* Other semantics could be decided at a later date.
* Similar to my [TaskManager](https://gist.github.com/ktbarrett/178cfed4f9963642eaf9ad27a3e32e16).

imphil, 2023-12-15:
Thanks for your excellent write-up, @ktbarrett. There's more research for me to do, but one question that came up in my head: why do we need symmetric tasks at all? Wouldn't asymmetric tasks (i.e., any spawned task is a child task) be sufficient and neatly lead to reasonable exception behavior, with no exception being swallowed: either an exception is caught within the test, otherwise it ends the test (like any other exception raised within the test itself).

marlonjames, 2023-12-15:
@imphil I think that is essentially what we have now.

The more I think about it, the more I think our current behavior is fine as the default.

If a user wants to capture the `Exception` that is raised, but not fail the test immediately, even if the task is not being joined on, they can opt-in to that if we provide some additional functionality.

Using `TaskGroup`s, we could provide an argument `cancel_on_exception=True` that when `False` stops the default behavior of cancelling all remaining tasks in the group on raising:

```py3
@cocotb.test
async def test_example(dut):
    # if Clock's start task throws an exception, the test should fail
    # the driver isn't working as intended so the test can't continue
    cocotb.start_soon(Clock().start())

    async with TaskGroup(cancel_on_exception=False) as tg:

        tg.start_soon(some_coro())

        @tg.start_soon
        async def do_stuff():
            await Timer(10, 'ns')
            raise ValueError

        # some_coro() Task will be cancelled if it is not finished

    # since "do_stuff" throws ValueError, it is reraised here
    # ends the test_example task with the ValueError
```

They would still have to handle any `Exception`s on exit of the context manager, but could defer it in this manner.

We could also add a `capture_exception=False` argument to `TaskGroup.start_soon()` that when `True` would accomplish the same thing, but scoped only to the task that is started, rather than the entire `TaskGroup`.

```py3
@cocotb.test
async def test_example(dut):
    # if Clock's start task throws an exception, the test should fail
    # the driver isn't working as intended so the test can't continue
    cocotb.start_soon(Clock().start())

    async with TaskGroup(cancel_on_exception=False) as tg:

        tg.start_soon(long_running_coro())

        # ValueError will be captured but not cancel long_running_coro() Task
        @tg.start_soon(capture_exception=True)
        async def do_stuff():
            await Timer(10, 'ns')
            raise ValueError

        # waits for all child tasks to finish

    # since "do_stuff" throws ValueError, it is reraised here
    # ends the test_example task with the ValueError
```

---

We could then extend this to the test level, by wrapping every test in a `TaskGroup`.

```py3
async def test_wrapper():
    with cocotb.scheduler.set_taskgroup(TaskGroup()):
        await test_example_task
```

And `cocotb.start_soon(coro, *, capture_exception=False)` would be forwarded by the scheduler to the wrapping `TaskGroup`.

I suppose in the case where a `capture_exception=True` task is created from the test, if that task has a captured `Exception` and is never joined when the test ends, it would fail the test.

ktbarrett, 2023-12-16:
Adding a parameter to decide what to do is essentially equivalent to separating them into two functions, I'm not opposed to that, but I'd guess that making ` cocotb.start_soon`'s default for  `capture_exception` `False` and  `TaskGroup.start_soon`'s `True` would be the best defaults.

TaskGroup ensure there is always a consistent location where child tasks' exceptions and results can be captured and  checked. This allows us to ensure that child tasks *never* exceed the lifetime of their parents; which to me, is the whole point. So things like `cancel_on_exception=False` defeat the purpose of the thing and prevent us from being able to guarantee that orphaned tasks with captured exceptions aren't ignored when they should be immediately and in a consistent location (hence the need to use a language structure such as context manager), handled.

marlonjames, 2023-12-16:
>  `TaskGroup.start_soon`'s `True` would be the best

This would mean the default in a `TaskGroup` is to **not** cancel the remaining tasks when a task raises an `Exception`. That's the opposite of asyncio and trio.

> things like `cancel_on_exception=False` defeat the purpose of the thing and prevent us from being able to guarantee that orphaned tasks with captured exceptions aren't ignored

I might not have been clear, but I don't think that my proposal does this. The only difference in my proposal is that when `capture_exception` is set to `True`, instead of cancelling the remaining tasks when an `Exception` is raised, it is "captured" until the end of the `TaskGroup`, whether because all of the other tasks finish or because a non-capture task raises, then it has to be handled at the exit of the `TaskGroup`. That's why I think `TaskGroup.start_soon` should have `capture_exception=False` by default, so that the original behavior is preserved.

ktbarrett, 2023-12-16:
The reason I deleted my comment was that I realized I was incorrect. It was too early in the morning to respond.