cocotb/cocotb #3591: Coverage support with ghdl-gcc simulator

EfesX, 2023-12-16:
I used ghdl-gcc to cover as follows:
```shell
ghdl -a -Wc,-fprofile-arcs -Wc,-ftest-coverage adder.vhd
ghdl -a adder_tb.vhd
ghdl -e -Wl,-lgcov adder_tb
ghdl -r adder_tb
```
But this is not possible with cocotb because the Ghdl(Simulator) class uses the call `ghdl -i` and then `ghdl -m` instead of `ghdl -a` and `ghdl -e`

<details>
<summary> log </summary>

```shell
INFO: Running command ghdl -i --work=top -Wc,-fprofile-arcs -Wc,-ftest-coverage /home/efesxzc/sandbox/adder.vhd in directory /home/efesxzc/sandbox/sim_build
ghdl:error: unknown warning identifier: c,-fprofile-arcs
Process 'ghdl' terminated with error 1

```
</details>

<details>
<summary> Ghdl(Simulator) </summary>

```python
class Ghdl(Simulator):
    supported_gpi_interfaces = {"vhdl": ["vpi"]}

    @staticmethod
    def _simulator_in_path() -> None:
        if shutil.which("ghdl") is None:
            raise SystemExit("ERROR: ghdl executable not found!")

    @staticmethod
    def _get_parameter_options(parameters: Mapping[str, object]) -> Command:
        return [f"-g{name}={value}" for name, value in parameters.items()]

    def _build_command(self) -> List[Command]:

        if self.verilog_sources:
            raise ValueError(
                f"{type(self).__qualname__}: Simulator does not support Verilog"
            )

        cmds = [
            ["ghdl", "-i"]
            + [f"--work={self.hdl_library}"]
            + self.build_args
            + [str(source_file) for source_file in self.vhdl_sources]
        ]

        if self.hdl_toplevel is not None:
            cmds += [
                ["ghdl", "-m"]
                + [f"--work={self.hdl_library}"]
                + self.build_args
                + [self.hdl_toplevel]
            ]

        return cmds

    def _test_command(self) -> List[Command]:

        cmds = [
            ["ghdl", "-r"]
            + [f"--work={self.hdl_toplevel_library}"]
            + self.test_args
            + [self.sim_hdl_toplevel]
            + ["--vpi=" + cocotb.config.lib_name_path("vpi", "ghdl")]
            + self.plusargs
            + self._get_parameter_options(self.parameters)
        ]

        return cmds
```
</details>

For example, creating a new GhdlGCC(Ghdl) class can solve this problem. But I'm not sure if this is the right way. 

<details>
<summary> GhdlGCC(Ghdl) </summary>

```python
class GhdlGCC(Ghdl):
        def _build_command(self) -> List[Command]:

        if self.verilog_sources:
            raise ValueError(
                f"{type(self).__qualname__}: Simulator does not support Verilog"
            )

        cmds = [
            ["ghdl", "-a"]
            + [f"--work={self.hdl_library}"]
            + self.build_args
            + [str(source_file) for source_file in self.vhdl_sources]
        ]

        return cmds

    def _test_command(self) -> List[Command]:
        if self.hdl_toplevel is not None:
            cmds = [
                ["ghdl", "-e"]
                + [f"--work={self.hdl_library}"]
                + self.elab_args
                + [self.hdl_toplevel]
            ]

            cmds += [
                ["ghdl", "-r"]
                + [f"--work={self.hdl_toplevel_library}"]
                + self.test_args
                + [self.sim_hdl_toplevel]
                + ["--vpi=" + cocotb.config.lib_name_path("vpi", "ghdl")]
                + self.plusargs
                + self._get_parameter_options(self.parameters)
            ]

        return cmds
```
</details>

imphil, 2023-12-16:
> But this is not possible with cocotb

More specificially, it's not possible with the cocotb-provided Makefiles for GHDL. You can call GHDL yourself and pass any command-line option, you only need to pass the right command-line options to load cocotb as well, as documented here: https://docs.cocotb.org/en/stable/custom_flows.html#ghdl

That being said, there has been some back and forth on the most appropriate way to call GHDL by default in our Makefiles. Please have a look at (closed) pull requests and issues, I don't remember the details. If you can come up with a solution that makes you and others happy, we're happy to accept a pull request!

EfesX, 2023-12-17:
Thanks, i got it.

closed 2023-12-17 by EfesX

EfesX, 2024-11-08:
Hi @jacob720 , I didn't work on this at cocotb. The solution for me was to use ghdl directly, like this:
```
ghdl -i adder.vhd
ghdl -m -Wc,-fprofile-arcs -Wc,-ftest-coverage -Wl,-lgcov adder
ghdl -r adder
lcov -c -d . -o coverage.info
genhtml -o html coverage.info
```

In cocotb, I have not found a way to pass the '-Wx, x' flags to 'ghdl -m' without passing them to 'ghdl -i'. But I didn't search well.

amellow, 2024-11-18:
A work around that I found is to first locate the Makefile for GHDL with:

`cocotb-config --makefiles`

Then, after going to the directory and opening Makefile.ghdl, locate the portion under the comment "# Compilation Phase"
From here, I modified the line for the '-m' command, adding in a new variable "$(ELAB_ARGS)" like so:

`$(CMD) -m $(GHDL_ARGS) $(ELAB_ARGS) $(COMPILE_ARGS) --workdir=$(SIM_BUILD) -P$(SIM_BUILD) --work=$(RTL_LIBRARY) $(TOPLEVEL)`

Then you can modify your project makefile to include this new variable, also you'll want your makefile to run the lcov and genhtml commands for you too. My makefile is shown below:

    SIM ?= ghdl
    TOPLEVEL_LANG ?= vhdl

    VHDL_SOURCES = ./adder.vhd

    ELAB_ARGS += -Wc,-fprofile-arcs -Wc,-ftest-coverage -Wl,-lgcov
    SIM_ARGS += --wave=dump.ghw --vcd=dump.vcd

    TOPLEVEL := adder
    MODULE := adder_tb

    include $(shell cocotb-config --makefiles)/Makefile.sim

    cover:
	    rm *e~*
	    lcov -c -d . -o coverage.info
	    genhtml -o html coverage.info