cocotb/cocotb #3597: Unable to locate command >vsim<.

hongyudeng2012, 2023-12-21:
set up:
cocotb 1.8.1 
64bit windows10
python 3.12.15... C:\ProgramData\Miniconda3\envs\cocotb_t3\python.exe
simulator:  C:/modelsim_dlx64_10.7a

test example code from https://github.com/cocotb/cocotb
dff.sv (no modification )
Makefile (add SIM = modelsim)
test_dff.py (no modification)

i have set up an envrionment in miniconda for cocotb. when i try to run the example. i get "unable to locate command vsim"
[image]
although C:\modelsim_dlx64_10.7a\win64pe  and C:\ProgramData\Miniconda3\envs\cocotb_t3\  are already in system variable.
i probabaly missing something simple.

i also try to modifiy the Makefile.questa in C:\ProgramData\Miniconda3\envs\cocotb_t3\Lib\site-packages\cocotb\share\makefiles\simulators by specifiy the path of modelsim, no help, still the same error  "unable to locate command vsim
[image]

cmarqu, 2023-12-21:
Something seems to be eating your path separators.
Maybe it would be easier to try https://github.com/themperek/cocotb-test 
or the experimental Python Runner (https://docs.cocotb.org/en/stable/runner.html),
or to write your own script using the info in https://docs.cocotb.org/en/stable/custom_flows.html#custom-flows

ktbarrett, 2023-12-21:
This has been reported in the Discussions (can't find which one) and on the Gitter a couple times now. I've seen it myself a couple years ago. I don't remember why this was happening or if there was any solution.

marlonjames, 2023-12-21:
It might be related to whatever the root cause of https://github.com/cocotb/cocotb/issues/3013 is.

hongyudeng2012, 2023-12-22:
> Something seems to be eating your path separators. Maybe it would be easier to try https://github.com/themperek/cocotb-test or the experimental Python Runner [...]
hi,cmarqu, thank you for your fast input. Yes, you are right. I guess it is some typical linux/windows format issue, namely / or \. But my knowledge of make and cygwin is not enough to get me a clue where.
I tried cocotb-test, the problem is gone. but I would still prefer to have it work in the classic way, because my existing project already works including all the CI configuration. if i switch to cocotb-test, everything has to start over again and i don't know maybe there are more issue. I still remember that i was not easy for me to config everything 3-4 years ago for the CI workflow.

hongyudeng2012, 2023-12-22:
> bork
hallo, ktbarrett, thank you for your thoughts.  could you please explain a little bit more on  "I'm thinking the mixing of conda's Python and the msys2 subsystem are just a bad idea." and what alternative i have?
bests

hongyudeng2012, 2023-12-22:
> I guess the real question is, how did it ever work in CI?

ah, it did work for me for many years.  Only until recently this problem occurs. could it possible related to windows update?

hongyudeng2012, 2024-01-03:
as i was afraid, i run into more issue with cocotb-test. Can someone tell me, on which platform is cocotb tested with modelsim?. Modelsim provide only 32bits for Linux. does it means that the released version of cocotb is tested through Modelsim_32bit in Linux?

themperek, 2024-01-03:
>Modelsim provide only 32bits for Linux. does it means that the released version of cocotb is tested through Modelsim_32bit in Linux?

No, it is not tested but it should work. https://github.com/cocotb/cocotb/wiki/Tier-2-Setup-Instructions#32-bit-python

I would disgorge using makefiles on Windows. It can work but ...

ktbarrett, 2024-01-04:
Yes, I recommend using Linux for testing with cocotb. You can even still use Windows while using Linux for cocotb testing by leveraging Windows Subsystem for Linux (WSL), which is what I use when working with cocotb at home.

hongyudeng2012, 2024-01-05:
> Yes, I recommend using Linux for testing with cocotb. You can even still use Windows while using Linux for cocotb testing by leveraging Windows Subsystem for Linux (WSL), which is what I use when working with cocotb (or anything RTL-related) at home.

I'm considering that as well. Do you use cocotb with modelsim as well? Modelsim only offer 32bit version for Linux. My concern is that I have to switch to python 32 which might cause more issue. do you have experience with that?

hongyudeng2012, 2024-01-05:
> > Modelsim provide only 32bits for Linux. does it means that the released version of cocotb is tested through Modelsim_32bit in Linux?
> 
> No, it is not tested but it should work. https://github.com/cocotb/cocotb/wiki/Tier-2-Setup-Instructions#32-bit-python
> 
> I would not recommend using makefiles on Windows. It can work but ...

thanks for your not recommendation. :)

closed 2024-02-26 by ktbarrett