cocotb/cocotb #3635: Unable to override icarus verilog language generation

TrAyZeN, 2024-01-04:
I am unable to override icarus verilog language generation by passing the `-g` flag in `COMPILE_ARGS`. It seems that icarus only keeps the last flag value.

The following command produced by the Makefile succeeds which is not expected if the language generation is overwritten.
`iverilog -o sim_build/sim.vvp -D COCOTB_SIM=1 -s alu -g 2001 -f sim_build/cmds.f -g2012 alu.v`

The following command fails with a syntax error which is expected if the language generation is overwritten.
`iverilog -o sim_build/sim.vvp -D COCOTB_SIM=1 -s alu -g2012 -g 2001 -f sim_build/cmds.f alu.v`

## Files to reproduce
alu.v
```verilog
enum {
    ADD
} operation;

module alu();

endmodule
```

Makefile
```makefile
SIM ?= icarus
TOPLEVEL_LANG ?= verilog

VERILOG_SOURCES += alu.v

TOPLEVEL = alu

MODULE = test_alu

COMPILE_ARGS = -g 2001

include $(shell cocotb-config --makefiles)/Makefile.sim
```

test_alu.py
```python
import cocotb

@cocotb.test()
async def test(dut):
    pass
```

## System information
- cocotb version: 1.8.0
- linux version: 6.6.9
- python version: 3.11.4
- icarus verilog version: 12.0

ktbarrett, 2024-01-04:
Involves removing the standard from [this line](https://github.com/cocotb/cocotb/blob/b71735f1ff8f188b6a513f7c20ac9f78fc4cfbd9/src/cocotb/share/makefiles/simulators/Makefile.icarus#L65) and [this line](https://github.com/cocotb/cocotb/blob/b71735f1ff8f188b6a513f7c20ac9f78fc4cfbd9/src/cocotb/runner.py#L555). Do we really *need* to set the standard here? If so, it might be a good idea to have the language standard be another variable which we default to `-g2012`, but let users override. Otherwise, we should just remove it and let users set it.

TrAyZeN, 2024-01-04:
Keeping the same API, the code in the [runner](https://github.com/cocotb/cocotb/blob/b71735f1ff8f188b6a513f7c20ac9f78fc4cfbd9/src/cocotb/runner.py#L555) lets users overwrite the language generation, as `-g2012` is specified before `build_args`. A potential fix for [Makefile.icarus](https://github.com/cocotb/cocotb/blob/b71735f1ff8f188b6a513f7c20ac9f78fc4cfbd9/src/cocotb/share/makefiles/simulators/Makefile.icarus#L65) would be to prepend the default to `COMPILE_ARGS` as follows:
```makefile
# iverilog seems to keep the last flag value specified, so -g2012 has to be prepended to let users overwrite it
COMPILE_ARGS := -g2012 $(COMPILE_ARGS) # Default to latest SystemVerilog standard
COMPILE_ARGS += -f $(SIM_BUILD)/cmds.f
```

ktbarrett, 2024-01-04:
Does #3637 look like it would solve your problem?

TrAyZeN, 2024-01-04:
Yes, it fixes my problem.

PR #3637 (2024-01-04): Move around commandline args in Makefile.icarus
Moves default language version in iverilog commandline args earlier to allow the user to override it later with COMPILE_ARGS. Closes #3635.
closed 2024-01-04 by ktbarrett