cocotb/cocotb #3639: Issue running make SIM=icarus WAVES=1

tthompsongmsi, 2024-01-04:
I've had fst working before. I re-installed my OS and now after installing cocotb and iverilog i can't run with fst option. I have zlib and zlib-devel installed.

Thank you in advance.

ktbarrett, 2024-01-04:
You are going to have to give more information than that. Please look at the notes when creating an issue.

vtguitar26, 2024-01-04:
Sorry I had everything in there and it looks like it didn't submit it. Must be where I placed that comment shown above. I will redo it tomorrow morning.

marlonjames, 2024-01-04:
I edited your initial post to remove the info from the  comment area

ktbarrett, 2024-01-04:
Lol. Guess I'll have to check for that next time.

vtguitar26, 2024-01-04:
Thank you marlonjames!

ktbarrett, 2024-01-04:
The support for FST is decided at compile time, [see the source](https://github.com/steveicarus/iverilog/blob/7c25e8506c5b9d91019760507a11cf5b18836018/vpi/sys_table.c#L48-L60). If you installed `libz` and `libz-devel` *after* compiling icarus, you'll continue to see the lack of support. Try rebuilding icarus now that they are installed.

Also, I don't know what this has to do with cocotb. If you have further issues please report them to the correct project https://github.com/steveicarus/iverilog.

closed 2024-01-04 by ktbarrett

vtguitar26, 2024-01-04:
Thank you for that information I did not know that about compilation time I did install those after installing iverilog.