cocotb/cocotb #3643: cocotb doesn't work with pipx without a workaround

relativityspace-jsmith, 2024-01-06:
I tried using pipx to install cocotb on a fresh Ubuntu 22.04 system, and it almost works, but has one issue.

For those who haven't heard of it, pipx is one of the most convenient ways to run Python apps on systems that block global pip package installation (incl. Ubuntu 22.04 and above).  It automatically handles creating the venv and wraps commands (like cocotb-config) so that they pass through to the venv automatically.

After installing the dependencies (e.g. Icarus and GTKWave) through apt-get, I tried
```
$ pipx install cocotb
$ make WAVES=1
```

This *almost* works, but the simulation fails with an import error:
```
rm -f results.xml
make -f Makefile results.xml
make[1]: Entering directory '/mnt/c/Users/jsmith/Documents/MyProject'
rm -f results.xml
MODULE=test_my_project TESTCASE= TOPLEVEL=MyProject TOPLEVEL_LANG=verilog \
         /usr/bin/vvp -M /home/jamie/.local/pipx/venvs/cocotb/lib/python3.8/site-packages/cocotb/libs -m libcocotbvpi_icarus   sim_build/sim.vvp -fst
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:76   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
Error: VVP input file 12.0 can not be run with run time version 10.3 (stable)
make[1]: *** [/home/jamie/.local/pipx/venvs/cocotb/lib/python3.8/site-packages/cocotb/share/makefiles/simulators/Makefile.icarus:95: results.xml] Error 1
make[1]: Leaving directory '/mnt/c/Users/jsmith/Documents/MyProject/MyProject/VerilogModel'
make: *** [/home/jamie/.local/pipx/venvs/cocotb/lib/python3.8/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2
```

It looks like what's going on is that the C++ code looks for the `VIRTUAL_ENV` environment variable to determine if a venv is being used, and pipx doesn't set this variable, so it erroneously concludes that a venv isn't being used, and fails to run properly.

If I run:
```
$ export VIRTUAL_ENV=$HOME/.local/pipx/venvs/cocotb/
```
then everything works!

imphil, 2024-01-06:
Do you know (or can figure out) why VIRTUAL_ENV is not set by pipx?

relativityspace-jsmith, 2024-01-06:
Hmm I didn't find anything originally but doing another search I found this issue: https://github.com/pypa/pipx/issues/887.  Looks like pipx doesn't do this as it's not part of the standard.
[https://opengraph.githubassets.com/435719528ada064af7dd08a2c22532cd27d8ab78c30a778a0816a2191574f7d5/pypa/pipx/issues/887]<https://github.com/pypa/pipx/issues/887>
VIRTUAL_ENV Environment Variable Missing · Issue #887 · pypa/pipx<https://github.com/pypa/pipx/issues/887>
Hello, I have an app that uses ansible's pip module to install certain python packages within the virtual environment, after its initial creation (when the app is run). Note that this works okay in...
github.com

WARNING: This message was sent by an external party. Send suspicious messages to ***@***.***

ktbarrett, 2024-01-06:
https://github.com/cocotb/cocotb/pull/615 introduced this. It sets the program name to the path of the python executable, which is not done when Pytho is embedded, so that `sys.path` is populated using paths relative to the Python executable.

From [the documentation on `Py_SetProgramName`](https://docs.python.org/3/c-api/init.html#c.Py_SetProgramName)

> It tells the interpreter the value of the argv[0] argument to the main() function of the program (converted to wide characters). This is used by [Py_GetPath()](https://docs.python.org/3/c-api/init.html#c.Py_GetPath) and some other functions below to find the Python run-time libraries relative to the interpreter executable.

The use of `VIRTUAL_ENV` is necessary as `sys.prefix` hasn't been initialized to the virtualenv's prefix yet and there really is no other way to get the exact executable path.

~~But do we need the exact executable path, or will using `Py_SetProgramName` with a generic executable name work? Or will embedding a generic python executable name (`python3.11`) during build using a define suffice?~~ No, some simulators screw with `PATH`.

Alternatively, we should investigate using `PYTHONHOME` to set the prefix instead, which IMO is the way embedded python was intended to be used.

ktbarrett, 2024-08-12:
Opened tracking issue #4091.

closed 2024-08-12 by ktbarrett