cocotb/cocotb #3648: Performance degredation accessing large memory arrays

apstrike, 2024-01-11:
There seems to be a large performance degradation when assigning to large memory arrays in CocoTB that’s non-linear to the growth of the array itself. 

This was tested with CocoTB v1.8.1 on Python3.7 (64 bit) with Questa 2021.2 (64-bit) as a simulator.

For the module below:

```
module test ();

localparam MLEM_L2 = 16;
localparam MLEM = 2 ** MLEM_L2;

logic clk = 0;
initial forever #1 clk = ~clk;

logic [27:0] mem [0:MLEM-1] = '{default : 'x};

endmodule
```

and the cocotb test module:

```
import cocotb
from cocotb.triggers import RisingEdge

@cocotb.test()
async def test(dut):

    mlen = len(dut.mem)
    dut.mem.value = [0]*mlen
    await RisingEdge(dut.clk)
```

As `MLEM_L2` increases, from 13 -> 16 the simulation -> real time ratio decreases much more than I’d expect. The results can be see in the table below:

| MLEM_L2 | SIM TIME (ns) | REAL TIME (s) | RATIO (ns/s) |
| -------- | ------------- | ------------- | ----------- |
| 13 | 1.00 | 0.72 | 1.38 |
| 14 | 1.00 | 2.82 | 0.36 |
| 15 | 1.00 | 10.87 | 0.09 |
| 16 | 1.00 | 84.89 | 0.01 |

ktbarrett, 2024-01-11:
Have you done any profiling to find out why this could be? And are you sure this is not an issue with the simulator?

apstrike, 2024-01-19:
I've tried it out with Verilator as the simulator and did not see the same kind of problem. I also ran with questa a wrapper file on top of the dut to emulate the assignment of the memory in the same way as was done in cocotb (but without using the VPI) and saw no performance degradation issues there. I think its a VPI issue but how to tell if its a problem on the CocoTB side of things or the questa side I'm not sure.  I'm also not sure how I could just test the Questa side of the VPI.

When I ran the test again with `MLEM_L2 = 16` and the cocotb profiler enabled (`COCOTB_ENABLE_PROFILING=1`) I get the following call graph, that might be useful. 
[image]

I don't fully understand it but it looks like there's a lot of calls to the GPI handler from `Scheduler._do_writes`.
Looking at the code of the function:
 ```
async def _do_writes(self):
     """An internal coroutine that performs pending writes"""
     while True:
         await self._writes_pending.wait()
         if self._mode != Scheduler._MODE_NORMAL:
             await self._next_time_step
 
          await self._read_write

         while self._write_calls:
             handle, (func, args) = self._write_calls.popitem(last=False)
             func(*args)
         self._writes_pending.clear()
```

It looks to me like we make a call to each handle in the `_write_calls` list. I wonder if its possible to only use one handle for a logic array (or maybe that's infeasible, pointless or extremely problematic) or somehow parallelize the write calls to each GPI handler?

ktbarrett, 2024-01-19:
> I wonder if its possible to only use one handle for a logic array

That is already done. Unless you are explicitly doing something different.

> somehow parallelize the write calls to each GPI handler?

The VPI, VHPI, FLI are probably not designed for that, and the GPI certainly isn't. Also there's no guarantee the same handle will appear in the list twice.

I wonder if the time reported in `set_signal_value_int` is weird due to how profiling is collected or if it's really taking that long, which would be truly absurd.

> I've tried it out with Verilator as the simulator and did not see the same kind of problem. I also ran with questa a wrapper file on top of the dut to emulate the assignment of the memory in the [...]

I feel that this means the problem is the simulator, not cocotb. I would imagine that some simulators implement large arrays as sparse arrays or minimal length vectors until they are used. I'm guessing every write (or N writes) is causing a reallocation and what you are doing is just hitting a performance corner case in Questa.

Does profiling other simulators show the same weird graph where 95% of the time is spend in `set_signal_value_int`?

apstrike, 2024-01-19:
> I feel that this means the problem is the simulator, not cocotb. I would imagine that some simulators implement large arrays as sparse arrays or minimal length vectors until they are used. I'm guessing every write (or N writes) is causing a reallocation and copy and what you are doing is just hitting a performance corner case in Questa.
> 

Right, I didn't really think about how the simulator might be implementing large arrays, I guess I assumed that it would be allocated at the start.

> Does profiling other simulators show the same weird graph where 95% of the time is spend in `set_signal_value_int`?

No it doesn't seem like it. At least when I look at the profiling with verilator as the simulator the graph is very different and only 1.5% of the time is spent in `set_signal_value_int`:

[image]

ok, so maybe it is just a performance issue with Questa then.

ktbarrett, 2024-01-19:
I'd definitely file a bug with Questa if you can. I feel like this isn't too odd of a case and it shouldn't be pessimising it.

ktbarrett, 2024-01-20:
Unless there is any other questions or issues, I'll close this. Feel free to reopen if needed.

closed 2024-01-20 by ktbarrett