cocotb/cocotb #3650: Add support for greenback

chevillotn, 2024-01-12:
The greenback library allows to re-enter into an asynchronous function from a synchronous function:
https://github.com/oremanj/greenback

This is very convenient to be able to call an existing library (with many internal calls) that will eventually end-up in trying to drive signals in the DUT.

In order to support `greenback`, `cocotb` needs to inform `sniffio` about its name:
scheduler.py:
```
    def _schedule(self, coroutine, trigger=None):
...
            coroutine._trigger = None
            prev_library, sniffio_library.name = sniffio_library.name, "cocotb"
            result = coroutine._advance(send_outcome)
            sniffio_library.name = prev_library
```

And also to provide a global function called `current_task()` such that `greenback can find out which task is currently running.

scheduler.py:
```
    def current_task(self):
        return self._current_task
```
__init__.py:
```
def current_task():
    return scheduler.current_task()
```

Using those very simple patches i was able to use `greenback` to re-enter asynch function and drive DUT signals.
I have not looked into corner cases though.

Would you consider having a look at supporting `greenback` in the future? That would be very useful.
Thanks a lot

marlonjames, 2024-01-12:
Take a look at `@cocotb.external` decorator, it has similar functionality. Each call to the external function is done on a separate thread. You can then make calls to an async function that is decorated with `@cocotb.function` that will run in the cocotb scheduler.

chevillotn, 2024-01-15:
I'm actually not sure the context is clearly understood.

I have a cocotb test which is asynch, calling a synch function which has many synch calls and ends up in a function that needs to drive DUT signals in asynch.

For example:
`async def a_test1():
    print('DBG: atest1')
    await Timer(10, units='ns')
    print('DBG: atest1 (delay)')

def test2():
    print('DBG: test2')
    return a_test1()

def test1():
    print('DBG: test1')
    return test2()

@cocotb.test()
async def test_cocotb(dut):
    test1()
`

I'm not sure how to apply the cocotb.external in this case. The only example i found in the cocotb source show a call from an async function to the external function. But this is absolutely not what i'm trying to do.

An example indeed would be beneficial.
Thanks

ktbarrett, 2024-01-15:
`cocotb.external` works in conjunction with `cocotb.function` (the names are terrible, yes). `cocotb.external` is applied to call a regular function from an cocotb (async) function *that will eventually call another cocotb (async) function*. `cocotb.function` is applied to call a cocotb (async) function from a regular function, *but only if that regular function was eventually called from a cocotb function*.

You can see the pair as a way to bridge the `async`/`await` callstack through regular functions.

```
async def a_test1():
    print('DBG: atest1')
    await Timer(10, units='ns')
    print('DBG: atest1 (delay)')

def test2():
    print('DBG: test2')
    return cocotb.function(a_test1)()

def test1():
    print('DBG: test1')
    return test2()

@cocotb.test()
async def test_cocotb(dut):
    await cocotb.external(test1)()
```

typically you'll see `cocotb.external` and `cocotb.function` applied as decorators, but I don't find that useful, so I would instead use the above example's syntax.

Also note there is no real checking on this. You can call a `cocotb.function` object from a non-`cocotb.external` context and IDK what will happen, but it's probably not what you want.

chevillotn, 2024-01-15:
Using this syntax seems to work on my side. Not obvious though, thanks for the example!

This works fine with the simple functions as above but when i try to use it with my nested functions i get the following error:
```
                                                            a = cocotb.external(self.async_ipbus.write_register)
                                                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                                                          File "/usr/local/lib/python3.11/dist-packages/cocotb/decorators.py", line 129, in __init__
                                                            self._log = SimLog(f"cocotb.external.{self._func.__qualname__}.{id(self)}")
                                                                                                  ^^^^^^^^^^^^^^^^^^^^^^^
                                                        AttributeError: 'external' object has no attribute '__qualname__'
```
which i'm not sure how to resolve. Any idea?

ktbarrett, 2024-01-15:
The error message implies you are wrapping the function in `external` twice? If so, don't do that. It's only needed once. Either use the decorator XOR use the above syntax.

chevillotn, 2024-01-16:
Hi,
Finally i managed to get it to work. There was a small issue on my side.
Thanks for the support i can now use external/function calls and have all i need to work.
There is no need to add support for greenback in this case so you can close the issue.

closed 2024-01-16 by chevillotn