cocotb/cocotb #3651: [bug/question?]

Febbe, 2024-01-12:
cocotb            1.8.1
cocotb-bus     0.2.1
fedora            6.6.8-100.fc38.x86_64
questa            21.3_2
python          3.11.6 (system)
 
Accessing `dut._sub_handles` results in an empty dictionary, even it has entries when I watch it via the debugger.
Later, at other places in the code, it suddenly works.
I can't figure out why this happens.
I also can access the handles when using `dut.my_sub_handle` directly.
What can be the cause of that?

ktbarrett, 2024-01-12:
`__dir__` calls `_discover_all`. Debuggers call `__dir__`. All of the above are private interfaces ans *will* change in 2.0, so please avoid using them.

closed 2024-01-12 by ktbarrett

Febbe, 2024-01-12:
Did you comment and close the wrong issue?
Or do you want to to tell me I shall not use debuggers?

ktbarrett, 2024-01-13:
So the intent is that what is documentated is what is public and all that isn't documented, is private. But the documentation is lacking. I've been doing some PRs to try to remedy that recently, so it should be more obvious in 2.0.

For 1.8, the public interface of `SimHandleBase` is:
- `_path`
- `_name`
- `_type`: this may change value in 2.0
- `_log`
- `_def_name`
- `_def_file`
- `get_definition_file`: removed in 2.0, use `_def_file` instead
- `get_definition_name`: removed in 2.0, use `_def_name` instead
- `hash(handle)`
- `repr(handle)`
- `len(handle)`
- `str(handle)`: removed in 2.0, use `handle._path` instead
- `handle == other_handle`

`RegionObject` adds
- `for sub_handle in handle: ...`

`HierarchyObject` adds
- `handle.sub_handle`
- `handle._id`: deprecated in 2.0, use `handle[sub_handle_name]` instead

`HierarchyArrayObject` adds
- `handle[sub_handle_idx]