cocotb/cocotb #3686: VPI does not find instance

teobiton, 2024-01-28:
Hi, I'm trying to move my project from cocotb 1.8.0 to cocotb 2.0.0. I have had errors that were easy to fix but got one that I don't know how to handle. I am using Verilator exclusively.

cocotb version: 2.0.0
verilator version: 5.006
os: Ubuntu 22.04.3 LTS
python: 3.10.12

```sh
[teob@hgserv tb] $ COCOTB_LOG_LEVEL=TRACE make
rm -f results.xml
"make" -f Makefile results.xml
make[1]: Entering directory '/home/teob/Documents/Projets/cryptopen/tb'
mkdir -p sim_build
/usr/local/bin/verilator -cc --exe -Mdir sim_build -DCOCOTB_SIM=1 --top-module sha1 --vpi --public-flat-rw --prefix Vtop -o Vtop -LDFLAGS "-Wl,-rpath,/home/teob/.local/lib/python3.10/site-packages/cocotb/libs -L/home/teob/.local/lib/python3.10/site-packages/cocotb/libs -lcocotbvpi_verilator" --trace --trace-structs --trace-fst --timescale 1ns/1ps /home/teob/Documents/Projets/cryptopen/hw/sha1/sha1.sv /home/teob/Documents/Projets/cryptopen/hw/sha1/sha1_core.sv /home/teob/Documents/Projets/cryptopen/hw/interface/simple_reg_interface.sv /home/teob/.local/lib/python3.10/site-packages/cocotb/share/lib/verilator/verilator.cpp
make -C sim_build  -f Vtop.mk
make[2]: Entering directory '/home/teob/Documents/Projets/cryptopen/tb/sim_build'
ccache g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -Os -c -o verilator.o /home/teob/.local/lib/python3.10/site-packages/cocotb/share/lib/verilator/verilator.cpp
ccache g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -Os -c -o verilated.o /usr/local/share/verilator/include/verilated.cpp
ccache g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -Os -c -o verilated_dpi.o /usr/local/share/verilator/include/verilated_dpi.cpp
ccache g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -Os -c -o verilated_vpi.o /usr/local/share/verilator/include/verilated_vpi.cpp
ccache g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -Os -c -o verilated_fst_c.o /usr/local/share/verilator/include/verilated_fst_c.cpp
ccache g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -Os -c -o verilated_threads.o /usr/local/share/verilator/include/verilated_threads.cpp
/usr/bin/python3 /usr/local/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vtop.cpp Vtop___024root__DepSet_h84412442__0.cpp Vtop___024root__DepSet_heccd7ead__0.cpp Vtop__Dpi.cpp Vtop__Trace__0.cpp Vtop__ConstPool_0.cpp Vtop___024root__Slow.cpp Vtop___024root__DepSet_h84412442__0__Slow.cpp Vtop___024root__DepSet_heccd7ead__0__Slow.cpp Vtop__Syms.cpp Vtop__Trace__0__Slow.cpp > Vtop__ALL.cpp
ccache g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -Os -c -o Vtop__ALL.o Vtop__ALL.cpp
echo "" > Vtop__ALL.verilator_deplist.tmp
Archive ar -rcs Vtop__ALL.a Vtop__ALL.o
g++    verilator.o verilated.o verilated_dpi.o verilated_vpi.o verilated_fst_c.o verilated_threads.o Vtop__ALL.a   -Wl,-rpath,/home/teob/.local/lib/python3.10/site-packages/cocotb/libs -L/home/teob/.local/lib/python3.10/site-packages/cocotb/libs -lcocotbvpi_verilator -lz  -pthread -lpthread -latomic   -o Vtop
rm Vtop__ALL.verilator_deplist.tmp
make[2]: Leaving directory '/home/teob/Documents/Projets/cryptopen/tb/sim_build'
rm -f results.xml
MODULE=test_tops TESTCASE= TOPLEVEL=sha1 TOPLEVEL_LANG=verilog \
         sim_build/Vtop 
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:76   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on Verilator version 5.006 2023-01-22
     0.00ns INFO     cocotb                             Running tests with cocotb v2.0.0.dev0+f7b37ca5 from /home/teob/.local/lib/python3.10/site-packages/cocotb
     0.00ns DEBUG    gpi                                Looking for packages over 1 implementations
     0.00ns WARNING  gpi                                vpi_iterate returned NULL for type vpiInstance for object NULL
     0.00ns DEBUG    gpi                                Reached end of iterator
     0.00ns INFO     cocotb                             Seeding Python random module with 1706442466
     0.00ns DEBUG    gpi                                Looking for root handle 'sha1' over 1 implementations
     0.00ns DEBUG    gpi                                Got a Root handle (sha1) back from VPI
     0.00ns DEBUG    gpi                                Checking sha1 exists
     0.00ns DEBUG    cocotb.regression                  Searching for tests in module test_tops
     0.00ns DEBUG    cocotb.regression                  Python Path: ,/usr/lib/python310.zip,/usr/lib/python3.10,/usr/lib/python3.10/lib-dynload,/home/teob/.local/lib/python3.10/site-packages,/usr/local/lib/python3.10/dist-packages,/usr/lib/python3/dist-packages
     0.00ns DEBUG    cocotb.regression                  PWD: /home/teob/Documents/Projets/cryptopen/tb
/home/teob/.local/lib/python3.10/site-packages/_pytest/assertion/rewrite.py:178: UserWarning: Python runners and associated APIs are an experimental feature and subject to change.
  exec(co, module.__dict__)
/home/teob/.local/lib/python3.10/site-packages/_pytest/assertion/rewrite.py:178: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  exec(co, module.__dict__)
/home/teob/.local/lib/python3.10/site-packages/_pytest/assertion/rewrite.py:178: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
  exec(co, module.__dict__)
     0.00ns DEBUG    gpi                                Searching for DataWidth
     0.00ns DEBUG    gpi                                Unable to map vpiConst type 0 onto GPI type
     0.00ns DEBUG    gpi                                VPI: DataWidth initialized with 32 elements
     0.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiParameter(41)
     0.00ns DEBUG    gpi                                Checking sha1.DataWidth exists
     0.00ns ERROR    cocotb                             cocotb testbench initialization failed. Exiting.
                                                        Traceback (most recent call last):
[... cut ...]
from cocotb.triggers import ClockCycles, RisingEdge, Timer

if cocotb.simulator.is_running():
    DATA_WIDTH = cocotb.top.DataWidth.value
    ADDR_WIDTH = cocotb.top.AddrWidth.value
    BLOCK_WIDTH = cocotb.top.BlockWidth.value
    BYTE_ALIGN = cocotb.top.ByteAlign.value
    DIGEST_WIDTH = cocotb.top.DigestWidth.value
```

The error seems to be related to the VPI that does not find the toplevel and objects attached to it. Am I missing something that was changed/removed from version 2.0.0 here ?

marlonjames, 2024-01-28:
This is your problem:
```
     0.00ns DEBUG    gpi                                Searching for DataWidth
     0.00ns DEBUG    gpi                                Unable to map vpiConst type 0 onto GPI type
```

Verilator's parameter support was not very complete until recently. Try either the latest release of v5.020 or build from Verilator's master branch.

marlonjames, 2024-01-28:
Closing this, feel free to open again if there is still an issue after trying newer Verilator.

closed 2024-01-28 by marlonjames

teobiton, 2024-01-29:
Hi, so I tried newer versions of Verilator: version 5.018 and 5.020 and I still have the same issue with these versions.
With the development version I get a totally different error on internal Verilator types with cocotb 1.8.1 and 2.0.0.

Just to be sure I tried again with cocotb version 1.8.1 and verilator 5.020  and the tests run as usual, so I still don't know what the problem is.  I'll investigate the GPI changes in cocotb later.

teobiton, 2024-02-07:
FYI, my issue was fixed by [a8f56c3](https://github.com/cocotb/cocotb/commit/a8f56c34b5a4ce571719d4a3aa2ebfd523d07e23)