cocotb/cocotb #3705: cocotb 1.8.1 + python 3.12.2 + activehdl 12 not working

diarmuidcwc, 2024-02-07:
i have been using cocotb for years. It is integrated into our design flow using the approach documented [here ](https://docs.cocotb.org/en/latest/custom_flows.html#aldec-active-hdl)

Currently we are on

- python 3.11.8 (32b)
- Active-HDL 12 (32b)
- cocotb 1.8.1
- 

This works without any issues.

When I upgrade to python 3.12.2, the simulation loads but hangs before it reports the version of simulator being run

Below is the output of the simulator in both the working and non-working scenario. I don't know if there's an option to produce a more verbose output but I'm willing to test any suggestions.

For now I will stick with python 3.11

`
#-------------------------------------------------------------------------------
# Open the simulator
#-------------------------------------------------------------------------------
asim -acdb -acdb_cov sb -acdb_file testp2.04_multiple_slots_24ch.acdb -gg_python=True -t 1ps -lib work  -gg_EpromCfgInitFile=../../../../../testbench/testp2.04_multiple_slots_24ch/cfg_eeprom.txt  -gg_EpromAuxInitFile=../../../../../testbench/testp2.04_multiple_slots_24ch/aux_eeprom.txt  testbench testbench -loadvhpi "D:/Jenkins_WS/AXN_ABM_401_Simulations/.venv/Lib/site-packages/cocotb/libs/cocotbvhpi_aldec.dll:vhpi_startup_routines_bootstrap" -interceptcoutput 
ELBREAD: Elaboration process.
ELBREAD: Elaboration time 0.1 [s].
KERNEL: Main thread initiated.
KERNEL: Kernel process initialization phase.
ELAB2: Elaboration final pass...
KERNEL: Warning: Some or all source files were compiled for Expression/Condition Coverage but Expression/Condition Coverage was not enabled at simulation initialization.
ELAB2: Create instances ...
KERNEL: Time resolution set to 1ps.
KERNEL: PLI/VHPI kernel's engine initialization done.
PLI: Loading library 'C:\Aldec\Active-HDL-12\bin\systf.dll'
VHPI: Loading library 'D:/Jenkins_WS/AXN_ABM_401_Simulations/.venv/Lib/site-packages/cocotb/libs/cocotbvhpi_aldec.dll'
ELAB2: Create instances complete.
SLP: Started
SLP: Elaboration phase ...
SLP: Elaboration phase ... done : 0.0 [s]
SLP: Generation phase ...
SLP: Generation phase ... done : 0.4 [s]
SLP: Finished : 0.5 [s]
SLP: 142 (4.81%) primitives and 6 (0.20%) other processes in SLP
SLP: 98 (31.41%) signals in SLP and 214 (68.59%) interface signals
ELAB2: Elaboration final pass complete - time: 1.3 [s].
ACDB: Code Coverage session started in hierarchical mode for local units.
KERNEL: SLP loading done - time: 0.0 [s].
KERNEL: SLP simulation initialization done - time: 0.0 [s].
KERNEL: Kernel process initialization done.
Allocation: Simulator allocated 402564 kB (elbread=265 elab2=400623 kernel=1675 sdf=0)
KERNEL: ASDB file was created in location D:\Jenkins_WS\AXN_ABM_401_Simulations\TICAD003\TICAD003\sim\activehdl\batch\testp2.04_multiple_slots_24ch\rundir\wave.asdb
#===============================================================================
# Setup
#===============================================================================
# Control messages logged. 
# 0 => Dont Ignore messages, 1 => Ignore Messages
set IgnoreError      0
set IgnoreFailure    0
set IgnoreNote       0
set IgnoreWarning    0
# Severity level an assertion must possess to stop the simulator. 
# 0=note, 1=warning, 2=error, 3=failure, 4=none
set breakassertlevel 3
# Severity level an assertion must possess to be reported in STDOUT
# 0=note, 1=warning, 2=error, 3=failure, 4=fatal error
set messagelevel     2
#===============================================================================
# Signals to trace
#===============================================================================
#trace -rec * 
# trace /testbench/gen_drivers__0/arinc429driver_INST/p_clk/real_bit_period
# trace /testbench/gen_drivers__0/arinc429driver_INST/p_clk/bit_period
[... 78 lines cut ...]
set IgnoreWarning    0
# Severity level an assertion must possess to stop the simulator.
# 0=note, 1=warning, 2=error, 3=failure, 4=none
set breakassertlevel 3
# Severity level an assertion must possess to be reported in STDOUT
# 0=note, 1=warning, 2=error, 3=failure, 4=fatal error
set messagelevel     2
#===============================================================================
# Signals to trace
#===============================================================================
#trace -rec *
# trace /testbench/gen_drivers__0/arinc429driver_INST/p_clk/real_bit_period
# trace /testbench/gen_drivers__0/arinc429driver_INST/p_clk/bit_period
# trace /testbench/gen_drivers__0/arinc429driver_INST/p_clk/half_bit_period
# trace /testbench/gen_drivers__0/arinc429driver_INST/p_clk/unf
# trace /testbench/gen_drivers__0/arinc429driver_INST/p_clk/rnd
#trace -rec /testbench/gen_drivers__0/arinc429driver_INST/*
force -deposit /testbench/TICAD003_INST/dbg_data_gen/msg_count "11111" 600us
transcript off
     -.--ns INFO     gpi                                ..mbed\gpi_embed.cpp:106  in set_program_name_in_venv        Using Python virtual environment interpreter at /cygdrive/c/ACRA/WORK/AXN_ABM_401/.venv\Scripts\python
     -.--ns INFO     gpi                                ..\gpi\GpiCommon.cpp:101  in gpi_print_registered_impl       VHPI registered
^CTerminate batch job (Y/N)? y
(.venv) [AXN_ABM_401] dcollins@DUB-1YDJBX3 ~/TICAD003/TICAD003
$
`

ktbarrett, 2024-02-26:
Any update on this? We don't have ActiveHDL available for debugging.

cmarqu, 2024-02-27:
Try running with https://docs.cocotb.org/en/stable/building.html#envvar-COCOTB_SCHEDULER_DEBUG set.

por3bski, 2024-02-27:
@ktbarrett 
If it helps, we are able to provide evaluation license for ActiveHDL.
Please contact louied@aldec.com to discuss details.

@diarmuidcwc 
Do you have possibility to share the code or check if problem occurs in the latest version of ActiveHDL ?

imphil, 2024-02-28:
>  If it helps, we are able to provide evaluation license for ActiveHDL.

I reached out to Louie to discuss this further.