cocotb/cocotb #3707: VHDL using generic width for a package type doesn't work

rramunni, 2024-02-07:
Cocotb version cocotb v2.0.0.dev0+a8f56c34
Ubuntu 22.04
GHDL version 4.0.0-dev (3.0.0.r823.g9e2b6fc95) [Dunoon edition]
Python 3.11.5 (Anaconda)

Trying to use a VHDL generic package in my dut. I can define package functions that use the generic, but when I define a package type using the generic as a width, the below error occurs. 

I've narrowed down a minimal reproducible example to the code below. I make some type that's an array of std_logic_vectors of width COUNTER_WIDTH. I then include this package in the dut architecture, passing the generic to it. I use the type in the dut and also include a function that uses the generic. The function works fine, but the type causes an error. I know this because I can replace the type definition with a hard-coded width and the test runs successfully.

This code is supported in ghdl, since I can compile and run a vhdl testbench with the dut and package in ghdl. Seems like cocotb has some sort of issue here, just wanted to know if this is supported or not.

Command line output:
```
ghdl -i  --std=08 --workdir=build --work=work ./counter_pkg.vhd ./counter.vhd && \
/home/rramunni/.local/bin/ghdl -m  --std=08 --workdir=build -Pbuild --work=work counter
analyze ./counter_pkg.vhd
analyze ./counter.vhd
elaborate counter
rm -f results/counter_tb.xml
MODULE=counter_tb TESTCASE= TOPLEVEL=counter TOPLEVEL_LANG=vhdl \
 /home/rramunni/.local/bin/ghdl -r   --workdir=build -Pbuild --work=work counter  --vpi=/home/rramunni/anaconda3/lib/python3.11/site-packages/cocotb/libs/libcocotbvpi_ghdl.so --wave=results/counter_tb.ghw --psl-report=results/counter_tb_psl.json --vpi-trace=results/counter_tb_vpi.log -gCOUNTER_WIDTH=8
loading VPI module '/home/rramunni/anaconda3/lib/python3.11/site-packages/cocotb/libs/libcocotbvpi_ghdl.so'
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:76   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
VPI module loaded!
make[1]: *** [/home/rramunni/anaconda3/lib/python3.11/site-packages/cocotb/share/makefiles/simulators/Makefile.ghdl:105: results/counter_tb.xml] Error 255
make[1]: Leaving directory '/home/rramunni/dev/sec-aes-ti2/tb/cocotb'
make: *** [/home/rramunni/anaconda3/lib/python3.11/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2
```

counter.vhd:
```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity counter is
    generic(
        COUNTER_WIDTH : positive
    );
    port(
        clk     : in std_logic;
        reset_n : in std_logic;
        count   : out unsigned(7 downto 0)
    );
end entity counter;

architecture rtl of counter is
    package p is new work.counter_pkg generic map (COUNTER_WIDTH => COUNTER_WIDTH);
    use p.all;

    signal r_1d_arr : byte_arr_1d_t;
    signal r_count : unsigned(COUNTER_WIDTH - 1 downto 0);
begin
    process(clk) is
    begin
        if rising_edge(clk) then
            if (reset_n = '0') then
                r_count <= (others => '0');
            else
                r_count <= r_count + 1;
            end if;
        end if;
    end process;

    r_1d_arr <= use_byte_arr_1d(std_logic_vector(r_count));

    count <= r_count when (r_1d_arr(0) = std_logic_vector(r_count)) else (others => '0');
end architecture rtl;
```

counter_pkg.vhd:
```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package counter_pkg is
    generic (COUNTER_WIDTH : positive);

    type byte_arr_1d_t is array(0 to 7) of std_logic_vector(COUNTER_WIDTH - 1 downto 0); -- works if you replace COUNTER_WIDTH with 8

    function use_byte_arr_1d(a: std_logic_vector(7 downto 0)) return byte_arr_1d_t;

end package counter_pkg;

package body counter_pkg is
    function use_byte_arr_1d(a : std_logic_vector(7 downto 0)) return byte_arr_1d_t is
        variable v_ret : byte_arr_1d_t := (others =>(others => '0'));
    begin
        for i in 0 to COUNTER_WIDTH - 1 loop
            v_ret(i) := a;
        end loop;
        return v_ret;
    end function use_byte_arr_1d;

end package body counter_pkg;
```

counter_tb.py:
```
import cocotb

from cocotb.clock import Clock
from cocotb.triggers import ClockCycles, FallingEdge

@cocotb.test()
async def no_count(dut):
    """Test no count if reset is 0"""
    cocotb.start_soon(Clock(dut.clk, 2, units="ns").start())
    dut.reset_n.value = 0
    await ClockCycles(dut.clk, 5)
    count = dut.count.value.integer
    assert count == 0
```

marlonjames, 2024-02-08:
> `--vpi-trace=results/counter_tb_vpi.log`

Could you provide this log?

Also, try running with `COCOTB_LOG_LEVEL=TRACE` to get more information about what is happening internal to cocotb's shared libraries.

rramunni, 2024-02-08:
The above was already ran with `COCOTB_LOG_LEVEL = TRACE`. Also the log file is completely empty.

marlonjames, 2024-02-08:
You could try setting [COCOTB_ATTACH](https://docs.cocotb.org/en/latest/building.html#envvar-COCOTB_ATTACH) and connecting with a debugger. It stops and waits for attach right before the log
`-.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered`

The very next thing should be registering for the VPI startup callback using `cbStartOfSimulation`.

rramunni, 2024-02-08:
here is the backtrace:
```
Program received signal SIGABRT, Aborted.
__pthread_kill_implementation (no_tid=0, signo=6, threadid=140289351952192) at ./nptl/pthread_kill.c:44
44      ./nptl/pthread_kill.c: No such file or directory.
(gdb) bt
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=140289351952192)
    at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=140289351952192) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=140289351952192, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007f97a9017476 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007f97a8ffd7f3 in __GI_abort () at ./stdlib/abort.c:79
#5  0x000055d30502c2aa in __gnat_last_chance_handler ()
    at /home/runner/work/ghdl/ghdl/build-gcc-12.1.0/../src/grt/grt-cgnatrts.c:23
#6  0x000055d305082098 in grt.waves.foreach_scalar_signal_number.range_pos_to_val (
    rti=0x55d3050f72e0 <std__standard__integer__BT__RTI>, rng=0x55d305d40b04, pos=0, val=...)
    at /home/runner/work/ghdl/ghdl/build-gcc-12.1.0/../src/grt/grt-rtis_utils.adb:293
#7  0x000055d30508234b in grt.waves.foreach_scalar_signal_number.pos_to_vstring (vstr=...,
    rti=0x55d3050f72e0 <std__standard__integer__BT__RTI>, rng=0x55d305d40b04, pos=0)
    at /home/runner/work/ghdl/ghdl/build-gcc-12.1.0/../src/grt/grt-rtis_utils.adb:334
#8  0x000055d305082643 in grt.waves.foreach_scalar_signal_number.handle_array_1 (
    arr_rti=0x55d3050fa960 <work__counter_pkg__byte_arr_1d_t__BT__RTI>, index=0)
    at /home/runner/work/ghdl/ghdl/build-gcc-12.1.0/../src/grt/grt-rtis_utils.adb:379
#9  0x000055d305082a9c in grt.waves.foreach_scalar_signal_number.handle_any (
    rti=0x55d3050fa9c0 <work__counter_pkg__byte_arr_1d_t__RTI>)
    at /home/runner/work/ghdl/ghdl/build-gcc-12.1.0/../src/grt/grt-rtis_utils.adb:454
#10 0x000055d305081fd2 in grt.waves.foreach_scalar_signal_number (ctxt=...,
    obj_type=0x55d3050fa9c0 <work__counter_pkg__byte_arr_1d_t__RTI>,
    obj_addr=(system.address) 0x55d305d40b98, is_sig=true, param=0)
    at /home/runner/work/ghdl/ghdl/build-gcc-12.1.0/../src/grt/grt-rtis_utils.adb:507
#11 0x000055d305085ab1 in grt.waves.write_signal_numbers (decl=...)
    at /home/runner/work/ghdl/ghdl/src/grt/grt-waves.adb:878
#12 0x000055d305085c44 in grt.waves.write_hierarchy_el (decl=...)
    at /home/runner/work/ghdl/ghdl/src/grt/grt-waves.adb:923
#13 0x000055d305085e18 in grt.waves.wave_put_hierarchy_1 (inst=..., step=step_hierarchy,
    match_list=0x0) at /home/runner/work/ghdl/ghdl/src/grt/grt-waves.adb:971
#14 0x000055d3050860a8 in grt.waves.wave_put_hierarchy_block (inst=..., step=step_hierarchy,
    match_list=0x0) at /home/runner/work/ghdl/ghdl/src/grt/grt-waves.adb:1043
#15 0x000055d3050861ed in grt.waves.wave_put_hierarchy (root=..., step=step_hierarchy)
    at /home/runner/work/ghdl/ghdl/src/grt/grt-waves.adb:1075
#16 0x000055d305088ab9 in grt.waves.write_hierarchy (root=...)
    at /home/runner/work/ghdl/ghdl/src/grt/grt-waves.adb:1670
#17 0x000055d305088f2a in grt.waves.wave_start ()
    at /home/runner/work/ghdl/ghdl/src/grt/grt-waves.adb:1755
#18 0x000055d305033840 in grt.hooks.call_start_hooks ()
    at /home/runner/work/ghdl/ghdl/src/grt/grt-hooks.adb:135
#19 0x000055d305012f40 in <__ghdl_simulation_init> ()
    at /home/runner/work/ghdl/ghdl/src/grt/grt-processes.adb:1181
#20 0x000055d30501332f in grt.processes.simulation ()
    at /home/runner/work/ghdl/ghdl/src/grt/grt-processes.adb:1331
#21 0x000055d30502c09a in __ghdl_run_through_longjump (
    func=0x55d305013322 <grt.processes.simulation>)
    at /home/runner/work/ghdl/ghdl/build-gcc-12.1.0/../src/grt/config/jumps.c:234
#22 0x000055d30502bc02 in grt.main.run_simul ()
    at /home/runner/work/ghdl/ghdl/src/grt/grt-main.adb:163
#23 0x000055d30502bce4 in grt.main.run () at /home/runner/work/ghdl/ghdl/src/grt/grt-main.adb:198
```

Somehow the value of Rng.I32.Dir in grt-rtis_utils.adb:293 is a number instead of dir_downto.

rramunni, 2024-02-14:
Found out this occurs because I had --wave in my SIM_ARGS instead of --vcd as custom types do not work with the VPI interface to cocotb.

themperek, 2024-02-15:
I close this. Feel free to reopen if necessary.

closed 2024-02-15 by themperek