cocotb/cocotb #3712: No root handle found in simple verilator testbench

prokie, 2024-02-13:
```
cocotb: 1.8.1
python: 3.10.12
verilator: 5.020 2024-01-01 rev v5.020-39-gd410e551f
WSL2 running Ubuntu 22.04.3 LTS
```
I have setup minimal testcase in [repo](https://github.com/prokie/verilator_tb) but I am getting the following error when I run make

```
(venv) pontus@pontus:/tmp/verilator_tb$ make
rm -f results.xml
make -f Makefile results.xml
make[1]: Entering directory '/tmp/verilator_tb'
mkdir -p sim_build
verilator -cc --exe -Mdir sim_build -DCOCOTB_SIM=1 --top-module tb_top --vpi --public-flat-rw --prefix Vtop -o Vtop -LDFLAGS "-Wl,-rpath,/tmp/verilator_tb/venv/lib/python3.10/site-packages/cocotb/libs -L/tmp/verilator_tb/venv/lib/python3.10/site-packages/cocotb/libs -lcocotbvpi_verilator" --timescale 1ns/1ps tb_top.sv /tmp/verilator_tb/venv/lib/python3.10/site-packages/cocotb/share/lib/verilator/verilator.cpp
make -C sim_build  -f Vtop.mk
make[2]: Entering directory '/tmp/verilator_tb/sim_build'
ccache g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable       -Os -c -o verilator.o /tmp/verilator_tb/venv/lib/python3.10/site-packages/cocotb/share/lib/verilator/verilator.cpp
ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable       -c -o verilated.o /usr/local/share/verilator/include/verilated.cpp
ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable       -c -o verilated_dpi.o /usr/local/share/verilator/include/verilated_dpi.cpp
ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable       -c -o verilated_vpi.o /usr/local/share/verilator/include/verilated_vpi.cpp
ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable       -c -o verilated_threads.o /usr/local/share/verilator/include/verilated_threads.cpp
/usr/bin/python3 /usr/local/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vtop.cpp Vtop___024root__DepSet_h84412442__0.cpp Vtop___024root__DepSet_heccd7ead__0.cpp Vtop__Dpi.cpp Vtop___024root__Slow.cpp Vtop___024root__DepSet_heccd7ead__0__Slow.cpp Vtop__Syms.cpp > Vtop__ALL.cpp
ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable       -c -o Vtop__ALL.o Vtop__ALL.cpp
echo "" > Vtop__ALL.verilator_deplist.tmp
Archive ar -rcs Vtop__ALL.a Vtop__ALL.o
g++ -fuse-ld=mold    verilator.o verilated.o verilated_dpi.o verilated_vpi.o verilated_threads.o Vtop__ALL.a   -Wl,-rpath,/tmp/verilator_tb/venv/lib/python3.10/site-packages/cocotb/libs -L/tmp/verilator_tb/venv/lib/python3.10/site-packages/cocotb/libs -lcocotbvpi_verilator  -pthread -lpthread -latomic   -o Vtop
rm Vtop__ALL.verilator_deplist.tmp
make[2]: Leaving directory '/tmp/verilator_tb/sim_build'
rm -f results.xml
MODULE=test_my_design TESTCASE= TOPLEVEL=tb_top TOPLEVEL_LANG=verilog \
         sim_build/Vtop 
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:105  in set_program_name_in_venv        Using Python virtual environment interpreter at /tmp/verilator_tb/venv/bin/python
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on Verilator version 5.020 2024-01-01
     0.00ns INFO     cocotb                             Running tests with cocotb v1.8.1 from /tmp/verilator_tb/venv/lib/python3.10/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1707843812
     0.00ns INFO     gpi                                Nothing visible via VPI
     0.00ns ERROR    gpi                                No root handle found
Traceback (most recent call last):
  File "/tmp/verilator_tb/venv/lib/python3.10/site-packages/cocotb/__init__.py", line 233, in _initialise_testbench
    _initialise_testbench_(argv_)
  File "/tmp/verilator_tb/venv/lib/python3.10/site-packages/cocotb/__init__.py", line 310, in _initialise_testbench_
    raise RuntimeError(f"Can not find root handle ({root_name})")
RuntimeError: Can not find root handle (tb_top)
Traceback (most recent call last):
  File "/tmp/verilator_tb/venv/lib/python3.10/site-packages/cocotb/__init__.py", line 332, in _sim_event
    scheduler.log.error(msg)
AttributeError: 'NoneType' object has no attribute 'log'
     0.00ns ERROR    gpi                                Passing event to upper layer failed
ERROR: results.xml was not written by the simulation!
make[1]: *** [/tmp/verilator_tb/venv/lib/python3.10/site-packages/cocotb/share/makefiles/simulators/Makefile.verilator:66: results.xml] Error 1
make[1]: Leaving directory '/tmp/verilator_tb'
make: *** [/tmp/verilator_tb/venv/lib/python3.10/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2
```

What am I doing wrong?

ktbarrett, 2024-02-13:
Your `tb_top` module is empty, verilator is optimizing the module out of existence. This is a known behavior of verilator (and Questa too).

prokie, 2024-02-13:
Aha, I only used Xcelium at work so I did not know this, not it works, thanks!

closed 2024-02-13 by prokie