cocotb/cocotb #3714: Release cocotb 1.9

imphil, 2024-02-14:
- [ ] Create stable/1.9 branch @imphil 
- [ ] Setup CI for stable/1.9 @imphil 
- [ ] Add backports through PRs.

ktbarrett, 2024-04-27:
@imphil we didn't add that. Also, instead of deprecating things in 2.0 we should remove them and deprecated them in 1.9.

- [ ] Add `FutureWarning` to `cocotb.runner` about the move.
- [ ] Add `FutureWarning` to `cocotb.config` about the move.
- [ ] Update `FutureWarning` in scheduler about the behavior of tasks getting exceptions
- [ ] Various deprecations in handle and triggers.

ktbarrett, 2024-06-20:
More backports
- [ ] #3951
- [ ] #3927

imphil, 2024-06-25:
- [ ] https://github.com/cocotb/cocotb/pull/3978

imphil, 2024-07-08:
@ktbarrett Are any of the remaining items in https://github.com/cocotb/cocotb/issues/3714#issuecomment-2080964264 critical? Otherwise I'm closing the 1.9 cycle now.

imphil, 2024-07-09:
1.9.0rc1 is now available at https://pypi.org/project/cocotb/1.9.0rc1/. I plan to release 1.9.0 final by Friday, unless something major comes up.

imphil, 2024-08-07:
cocotb 1.9 is released.

closed 2024-08-07 by imphil