cocotb/cocotb #3722: deprecate `stage` in cocotb tests/RegressionManager

ktbarrett, 2024-02-19:
This feature existed to order tests logically into stages: tests from earlier stages are run before tests from later stages. But no additionally functionality wrt to this was added. Did a regression pass a stage? What does it mean if anything for it to pass a stage? Why can't the user just reorder tests lexically? Why does test order matter? Shouldn't they be independent?

So, I feel this should be deprecated without an alternative and removed in 3.0.

cmarqu, 2024-02-19:
FWIW, this came in via https://github.com/cocotb/cocotb/commit/a465523e6197b096b41e281cae6256dd61e9b997

ktbarrett, 2024-02-19:
> This can be used to ease the analysis of output files or the test outputs.

Any idea what's meant by this?

cmarqu, 2024-02-19:
Ping @wallento

ktbarrett, 2024-12-22:
I suppose this isn't hurting anything.

closed 2024-12-22 by ktbarrett