cocotb/cocotb #3724: Useful Feature: Separating Python side from simulation side

muffgaga, 2024-02-19:
We would like to be able to use `cocotb` in decoupled testbench/simulator software environments.
(Background: our Python-wrapped (containerized) software stack is difficult to build on the old RHEL versions our simulator requires.)

Since there seems to be a very limited set of C functions that are linked to the simulator (those below the "Generic Language Interface"), we thought about adding a remote procedure call interface that decouples this layer into a client-side "testbench" (or rather the Python parts of `cocotb` and a bit of the C/C++ to get up to the actual call to RPC) and a server-side "simulator".
With such an RPC layer, we could run the client and server sides in separate software environments, or even on different machines depending on the RPC mechanism.
(In the past, we have used a C++-based RPC library to implement this for our custom DPI-coupled simulator interface that focuses on higher-level/integration testing.)

Might such a change be of interest to other parts of the community, or are there other suggestions that we should consider?

ktbarrett, 2024-02-19:
It's definitely of interest. We (the maintainers, power users, previous personal employers, etc.) have been tossing the idea around for a couple years now, but no one has had time to work on it.

Primarily we are interested in it to be able to have VPI and DPI functionality usable from the same testing environment. We could also vastly simplify and improve the cocotb codebase by leveraging existing tools like pytest for regressions and asyncio or trio for concurrency.

The primary concern is performance. Doing an RPC 4 times for every clock cycle to implement a clock driver in Python would kill. But obviously, that could be alleviated by moving certain things to HDL.

Considering the massive changes this would involve, I don't see this is as a future for cocotb, but instead a successor/alternative.

muffgaga, 2024-02-20:
Nice to read that and thank you for the detailed answer :)

Regarding your performance concern:
Are there any examples/tests that could serve as a benchmark? We probably could hack something that could provide some insights into the overhead that we might have to expect — for possible clean implementation I would also see this kind of feature as an "option" to activate, i.e. not the default mode of operation.
I do agree that RPC via some network could be really problematic when performed at each clock cycle (been there ;) ), but I guess a machine-local IPC mechanism would be still very much feasible (if it's about call latency we could resort to shmem and futexes, that should keep the call latency way below us range; i.e. I would expect that anything below 1MHz VPI/DPI call rate would be quite possible.

cmarqu, 2024-02-20:
The benchmark we have so far is https://github.com/search?q=repo%3Acocotb%2Fcocotb%20benchmark&type=code, with https://cocotb.github.io/cocotb-benchmark-results/dev/bench/ being the visualization for it.

For such a big change though, a more real-life testbench like the one for https://github.com/corundum/corundum/ might be better.