cocotb/cocotb #3725: Support bluesim as Simulator backend

Febbe, 2024-02-20:
It would be very usefull, to add bluesim as simulator, since it compiles the bsv code directly into an executable (bsv → c++ → asm). That increases the simulation speed, which is currently the largest bottleneck, simulating designs.

ktbarrett, 2024-02-20:
Does bluesim implement the VPI or a similar procedural interface?

Febbe, 2024-02-21:
Unfortunately I am not that deep into bluesim. 

From my understanding of chapter 5 in [https://github.com/B-Lang-org/bsc/releases/latest/download/bsc_user_guide.pdf] and chapter 2.5.3, the bsc chooses a similar approach like cocotb, by calling pure verilog functions via VPI and a corresponding simulator, but the bluespec code is translated directly to executable code.

cocotb could link to bluesim directly without the VPI as layer in between.
There is also a proposal to use cocotb as layer to other simulators, instead of creating their own:
https://github.com/B-Lang-org/bsc/issues/170#issuecomment-646031335

Therefore, I am afraid that, the VPI support of bluespec is not compatible to cocotb, and to support bluesim, you would've to implement a direct connection to bluesim via runtime linking. Bluesim would then communicate itself via VPI with the simulator for verilog.

ktbarrett, 2024-02-21:
AFAICT bluespec is just a transpiler? So it should be possible to have the transpilation stage be run before the cocotb simulation stage is run. What is needed from cocotb here? Gentle reminder cocotb is not a project automation framework, so unless there is something special necessary to support bluespec, I'd consider this out of scope.

Febbe, 2024-02-22:
Not directly. [Bluespec is the high level HDL.](https://github.com/B-Lang-org/bsc) And bsc combines the compiler(transpiler to verilog), simulator (transpiler to C++ and an executable) and tools.

The "bluesim" simulator generator is executed via `bsc -sim ...` and generates both a simulator executable and a linkable library. 

What I would like to see in cocotb is, that cocotb uses the **generated** simulator executable or the linkable library as generator backend, which would be way faster, regarding simulation speed.

marlonjames, 2024-02-22:
You can take a look at Verilator and how they implement some of the VPI specification.
If bluesim were to add an implementation of VPI to the simulator that supported the parts that cocotb needs, then it could work.