cocotb/cocotb #3741: [Feature Request] Add Trigger to the with_timeout function

Febbe, 2024-02-27:
When having a time-unit based timed clk or a subclk or any signal with non deterministic trigger behaviour, using the  `cocotb.triggers.with_timeout` coroutine is not a feasible solution. 

But it would be an expected usecase for all triggers and coroutines, which should only run for a specific amount of RisingEdges of another handle.

It would also allow to write tests with a fail condition aka. Run test X until 100 RisingEdges of the AXI rdy handle arise.
Using a time-unit based  or step based timer may not be usable here, since the other end of the AXI may block progress due to contention.

ktbarrett, 2024-02-27:
Does `First` not work for you? `with_timeout` is just a convenience wrapper around `First`.

Febbe, 2024-02-27:
> Does `First` not work for you? `with_timeout` is just a convenience wrapper around `First`.

Yes, I wrote my own.