cocotb/cocotb #3742: Possible Callbacks not cleaned up after cancelation of coro

Febbe, 2024-02-27:
The cancelation of an coroutine task results in the errorlog below.
I am using 
```
cocotb          1.8.1
cocotb-bus      0.2.1
pyuvm           2.9.1
```

A minimal example to trigger this might be 
```
fill up
```

# 5071900.00ns ERROR    ..est-framework/scoreboard.py(151) [uvm_test_top.env.scoreboard]: Timeout reached. Canceled test after 1000 clk cycles!
# timeout! raise exception
# 5071900.00ns ERROR    ..otb.Task 74.Scoreboard.run_phase Exception raised by this forked coroutine
# ** Warning: (vsim-3116) Problem reading symbols from /lib64/librt.so.1 : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /lib64/libdl.so.2 : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /lib64/libm.so.6 : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /lib64/libpthread.so.0 : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /lib64/libc.so.6 : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /lib64/ld-linux-x86-64.so.2 : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /lib64/libnss_sss.so.2 : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /scratch/fk68vatu/mypyenv/venv/lib64/python3.11/site-packages/cocotb/libs/libcocotbvpi_modelsim.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /scratch/fk68vatu/mypyenv/venv/lib64/python3.11/site-packages/cocotb/libs/libgpi.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /scratch/fk68vatu/mypyenv/venv/lib64/python3.11/site-packages/cocotb/libs/libgpilog.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /lib64/libgcc_s.so.1 : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /scratch/fk68vatu/mypyenv/venv/lib64/python3.11/site-packages/cocotb/libs/libcocotbutils.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /scratch/fk68vatu/mypyenv/venv/lib64/python3.11/site-packages/cocotb/libs/libembed.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /usr/lib64/libpython3.11.so.1.0 : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /scratch/fk68vatu/mypyenv/venv/lib64/python3.11/site-packages/cocotb/libs/libcocotb.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /scratch/fk68vatu/mypyenv/venv/lib64/python3.11/site-packages/cocotb/libs/libpygpilog.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /usr/lib64/python3.11/lib-dynload/_typing.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /usr/lib64/python3.11/lib-dynload/math.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /usr/lib64/python3.11/lib-dynload/_bisect.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /usr/lib64/python3.11/lib-dynload/_random.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /usr/lib64/python3.11/lib-dynload/_sha512.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /usr/lib64/python3.11/lib-dynload/_ctypes.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /lib64/libffi.so.8 : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /usr/lib64/python3.11/lib-dynload/_struct.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /scratch/fk68vatu/mypyenv/venv/lib64/python3.11/site-packages/cocotb/simulator.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /usr/lib64/python3.11/lib-dynload/_opcode.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /usr/lib64/python3.11/lib-dynload/_decimal.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /lib64/libmpdec.so.3 : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /usr/lib64/python3.11/lib-dynload/_hashlib.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /lib64/libcrypto.so.3 : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /lib64/libz.so.1 : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /usr/lib64/python3.11/lib-dynload/_blake2.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
# vsim_stacktrace.vstf written
# Current time Tue Feb 27 14:20:19 2024
# Program = vsim
# Id = "2019.4"
# Version = "2019.10"
# Date = "Oct 15 2019"
# Platform = "linux_x86_64"
# Signature = 039f21da4bb6a46c08c149b1c8122753
# 0    0x00000000004ebe79: '<unknown (@0x4ebe79)>'
# 1    0x000000000060cd28: '<unknown (@0x60cd28)>'
# 2    0x000000000060d7d7: '<unknown (@0x60d7d7)>'
# 3    0x0000000000611264: '<unknown (@0x611264)>'
# 4    0x000000000103fa75: '<unknown (@0x103fa75)>'
# 5    0x000000000060a253: '<unknown (@0x60a253)>'
# 6    0x0000000000785d60: '<unknown (@0x785d60)>'
# 7    0x00000000004f5e25: '<unknown (@0x4f5e25)>'
# 8    0x000000000068d1d3: '<unknown (@0x68d1d3)>'
# 9    0x0000000000b97c63: '<unknown (@0xb97c63)>'
# 10   0x0000000000b9c5ab: '<unknown (@0xb9c5ab)>'
# 11   0x0000000000b9e2de: '<unknown (@0xb9e2de)>'
# 12   0x0000000000e58eed: '<unknown (@0xe58eed)>'
# 13   0x0000000001593e5d: '<unknown (@0x1593e5d)>'
[... 35 lines cut ...]
# ** Warning: (vsim-3116) Problem reading symbols from /usr/lib64/python3.11/lib-dynload/_lzma.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /lib64/liblzma.so.5 : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /usr/lib64/python3.11/lib-dynload/unicodedata.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /usr/lib64/python3.11/lib-dynload/_uuid.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /lib64/libuuid.so.1 : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /usr/lib64/python3.11/lib-dynload/_csv.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /usr/lib64/python3.11/lib-dynload/_datetime.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /usr/lib64/python3.11/lib-dynload/_json.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /usr/lib64/python3.11/lib-dynload/_queue.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /usr/lib64/python3.11/lib-dynload/termios.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /usr/lib64/python3.11/lib-dynload/_pickle.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
# ** Warning: (vsim-3116) Problem reading symbols from /scratch/fk68vatu/mypyenv/venv/lib64/python3.11/site-packages/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_cython.cpython-311-x86_64-linux-gnu.so : module was loaded at an absolute address.
[... cut ...]
make[1]: *** [/scratch/fk68vatu/mypyenv/venv/lib64/python3.11/site-packages/cocotb/share/makefiles/simulators/Makefile.questa:160: results.xml] Error 211
make[1]: Leaving directory '/scratch/fk68vatu/dexie-stimulus/fk_pyuvm/test-framework'
make: *** [/scratch/fk68vatu/mypyenv/venv/lib64/python3.11/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2
```

Look like Questas handles are not updated just after the cancelation of a task.

_Originally posted by @Febbe in https://github.com/cocotb/cocotb/issues/1688#issuecomment-1966542114_

ktbarrett, 2024-02-27:
> Look like Questas handles are not updated just after the cancelation of a task.

Could you explain what you mean here? Certainly the stacktrace is unexpected.

Febbe, 2024-02-27:
> > Look like Questas handles are not updated just after the cancelation of a task.
> 
> Could you explain what you mean here? Certainly the stacktrace is unexpected.

For me it looks like, some function handle is called which does not exist anymore in the process of vsimk.
Therefore it crashes with SIGSEG. And from my understanding cocotb registers callbacks via VPI. 
My assumption is, that those handles are not cleaned up completely after a coroutine which references those is canceled.
Upon the resume of the simulation, vsimk tries to call these, but they are already cleaned by pythons GC.

Febbe, 2024-02-27:
I think its a duplicate of https://github.com/cocotb/cocotb/issues/1261, I just triggered it via the cancelation of a coroutine.

Febbe, 2024-02-27:
False alarm, removed some time-consuming and inefficient code and it crashes again. Even with a timeout of a second.

Febbe, 2024-02-27:
Boiled it down to those [lines](https://github.com/cocotb/cocotb/blob/393b3284b05a27b4c7e8c0d6635492347a7f03fe/src/cocotb/triggers.py#L890C1-L894C21) in cocotb:

```
        # kill all the other waiters
        # TODO: Should this kill the coroutines behind any Join triggers?
        # Right now it does not.
        for w in waiters:
            w.kill()
```            

And to answer the TODO: Yes, `First` should also kill all coroutines behind Join. Since that's what everyone would expect, when wrapping a coroutine in a task via the Join trigger.

ktbarrett, 2024-02-27:
@Febbe that's been a point of contention for a while and also an interface breaking change. I suggested we instead replace `First` and `Combine` in #2659. But regardless of that, no stacktrace should be occuring.

Febbe, 2024-02-27:
I have a new reproducible example:
```
import cocotb
from cocotb.triggers import RisingEdge, FallingEdge, ClockCycles 
from cocotb.clock import Clock

async def cicle_timer(sig, cicles):
    for _ in range(0, cicles):
        await RisingEdge(sig)
        await FallingEdge(sig)  
    return cicle_timer(sig, cicles)

async def wrapClockCycles(sig, cicles):
    await ClockCycles(sig, cicles)

async def with_timout(awaitable, awaitable_timout):
    print("started await with timeout")
    async def throw_timeout(awaitable_timout):
        print("await end in throw_timeout")
        await awaitable_timout
        print("timeout! raise exception")        
        raise cocotb.result.SimTimeoutError()         
    
    awt = cocotb.start_soon(awaitable)
    awtt = cocotb.start_soon(throw_timeout(awaitable_timout))
    return await cocotb.triggers.First(awt, awtt)

@cocotb.test()
async def test_crash(dut):
    c = Clock(dut.clk, 10, 'ns')
    await cocotb.start(c.start())
    # await with_timout(cicle_timer(dut.clk, 1000), cicle_timer(dut.clk, 100))
    # await with_timout(wrapClockCycles(dut.clk, 1000), wrapClockCycles(dut.clk, 100))
    await with_timout(wrapClockCycles(dut.clk, 1000), cicle_timer(dut.clk, 100))
```

The interesting thing is, when I use either of the commented lines, it works.
But as soon I use the last line, it crashes.
The DUT is relatively large, simulating a complete RISCV-piccolo32 processor with some extra extensions.

When I comment, the `#await FallingEdge(sig)` it does not crash:

```
async def cicle_timer2(sig, cicles):
    for _ in range(0, cicles):
        await RisingEdge(sig)
        #await FallingEdge(sig)
```

Adding a `RisingEdge(sig)` will not crash. So as soon the awaitable ends on a low clk it crashes.

```
async def cicle_timer2(sig, cicles):
    for _ in range(0, cicles):
        await RisingEdge(sig)
        await RisingEdge(sig)
```

ktbarrett, 2024-02-28:
I wonder if this is more of the same as #3010. There are notes all over the VPI about "Modelsim doesn't like this so we do something awful lol". The way that handles are cleaned up in the GPI is confusing, so I wouldn't doubt something wrong is done there.

Febbe, 2024-02-29:
Mhh, when #3010 is the related issue, it might be an easy but dirty fix, to hold a reference to the callbacks in a context manager and flag those as removed. You could add some debug code there to find out why the resource was not removed from vsimk earlier. At least it would not crash anymore, since the handler still exist. I assume, that the coroutine still exist, and the triggers lost any lifetime handle, therefore the GC removed them with their respective callbacks.

ktbarrett, 2024-03-01:
Questa version is 2019.10 per the error message. It might be worth seeing if a newer version of the simulator prevents the stacktrace, otherwise at some point someone will go about trying to debug this.

Febbe, 2024-03-02:
I can try a newer version and check if this still appears there. Just won't be using it for production, as my supervisor told me that it's even slower than Questa 2019.

ktbarrett, 2025-03-25:
So we did a lot of work on cleaning up GPI handles on cancellation (#4392), so this *may* be resolved now. Of course Questa 2019 is very old at this point, so it might not have the same VPI capability to make that change work. We don't officially support version of Questa that old.

One reason for the slowdown could be due to how they made radical changes to the optimizer in 2020 and it regressed in performance when using cocotb because of the access requirements. Newer versions of Questa got better with optimizing while supporting increased visibility for cocotb. So I'd definitely try a newer version of Questa at this point.

closed 2025-05-02 by github-actions[bot]