cocotb/cocotb #3743: Can't store value in array

juliusgodo, 2024-02-27:
I am using
```
cocotb         1.8.1
Python         3.11.6
```
and GHDL as simulator.

When trying to store values inside a array the value is not actually stored. Storing values in "normal" signals works.

VHDL signals:

```
type store_t is array (0 to 4) of unsigned(31 downto 0);
signal testArray: store_t := (others => x"F0F0F0F0");
signal testVec : std_logic_vector(31 downto 0);
```

cocotb test:
```
@cocotb.test()
async def hardwaretest(dut):
	dut.testarray[0].value = 15
	dut.testVec.value = 1337

	await Timer(50, "ns")
	
	print(dut.testarray[0].value)
	print(dut.testVec.value)
	
        assert int(dut.testVec.value) == 1337
	assert int(dut.testarray[0].value) == 15
```

results in:
```
ghdl -r -fsynopsys --std=93 --syn-binding --workdir=work procTest --wave=hardwaretest.ghw --ieee-asserts=disable-at-0 --vpi-trace=hardwaretest.vpi_trace.txt --vpi=$(cocotb-config --lib-name-path vpi ghdl)
loading VPI module '<pathToModule>/libcocotbvpi_ghdl.so'
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:105  in set_program_name_in_venv        Using Python virtual environment interpreter at <pathToVenv>/bin/python
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
VPI module loaded!
     0.00ns INFO     cocotb                             Running on GHDL version 4.0.0-dev (3.0.0.r871.g45fcff5bc) [Dunoon edition]
     0.00ns INFO     cocotb                             Running tests with cocotb v1.8.1 from <pathToCocotb>/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1709062477
     0.00ns INFO     cocotb.regression                  Found test integrationTests.hardwaretest2
     0.00ns INFO     cocotb.regression                  running hardwaretest(1/1)
    50.00ns INFO     cocotb.regression                  hardwaretest failed
                                                        Traceback (most recent call last):
                                                          File "<pathToFile>", line 102, in hardwaretest
                                                            assert int(dut.testarray[0].value) == 15
                                                        AssertionError: assert 4042322160 == 15
                                                         +  where 4042322160 = int(11110000111100001111000011110000)
                                                         +    where 11110000111100001111000011110000 = ModifiableObject(proctest.testarray[0]).value
    50.00ns INFO     cocotb.regression                  ****************************************************************************************
                                                        ** TEST                            STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
                                                        ****************************************************************************************
                                                        ** integrationTests.hardwaretest2   FAIL          50.00           0.01       8342.23  **
                                                        ****************************************************************************************
                                                        ** TESTS=1 PASS=0 FAIL=1 SKIP=0                   50.00           0.04       1427.85  **
                                                        ****************************************************************************************
```

When looking at the VPI-trace:
```
vpi_register_cb ({reason=cbStartOfSimulation, obj=0000000000000000}) = 00005F1F006EF600
vpi_register_cb ({reason=cbEndOfSimulation, obj=0000000000000000}) = 00005F1F00713750
vpi call callback 00005F1F006EF600 cbStartOfSimulation
 vpi_get_vlog_info
 vpi_get_vlog_info
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 0ms
 vpi_get (vpiTimePrecision, 0000000000000000) = -15
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 0ms
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 0ms
 vpi_iterate (vpiModule, 0000000000000000) = 00005F1F00D11EA0
 vpi_scan (00005F1F00D11EA0) = 00005F1F00D90520
 vpi_get (vpiType, 00005F1F00D90520) = vpiModule
 vpi_free_object (00005F1F00D11EA0)
 vpi_get_str (vpiFullName, 00005F1F00D90520) = "proctest"
 vpi_get (vpiType, 00005F1F00D90520) = vpiModule
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 0ms
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 0ms
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 0ms
 vpi_handle_by_name ("proctest.testarray", 0000000000000000) = 00005F1F00D11EA0
 vpi_get (vpiType, 00005F1F00D11EA0) = vpiNetArray
 vpi_get (vpiType, 00005F1F00D11EA0) = vpiNetArray
 vpi_get_str (vpiName, 00005F1F00D11EA0) = "testarray"
 vpi_iterate (vpiRange, 00005F1F00D11EA0) = 0000000000000000
 vpi_handle (vpiLeftRange, 00005F1F00D11EA0) = 00005F1F00D5AF90
 vpi_get_value (00005F1F00D5AF90, {format=Int}) = 0
 vpi_handle (vpiRightRange, 00005F1F00D11EA0) = 00005F1F00D42170
 vpi_get_value (00005F1F00D42170, {format=Int}) = 4
 vpi_get_str (vpiType, 00005F1F00D11EA0) = "???"
 vpi_handle_by_index (00005F1F00D11EA0, 0) = 00005F1F00C88710
 vpi_get (vpiType, 00005F1F00C88710) = vpiNet
 vpi_get (vpiType, 00005F1F00C88710) = vpiNet
 vpi_get (vpiSize, 00005F1F00C88710) = 32
 vpi_get (vpiVector, 00005F1F00C88710) = 1
 vpi_iterate (vpiRange, 00005F1F00C88710) = 0000000000000000
 vpi_handle (vpiLeftRange, 00005F1F00C88710) = 00005F1F00C49740
 vpi_get_value (00005F1F00C49740, {format=Int}) = 31
 vpi_handle (vpiRightRange, 00005F1F00C88710) = 00005F1F00C56190
 vpi_get_value (00005F1F00C56190, {format=Int}) = 0
 vpi_get_str (vpiType, 00005F1F00C88710) = "???"
 vpi_handle_by_name ("proctest.testVec", 0000000000000000) = 00005F1F00D2A2A0
 vpi_get (vpiType, 00005F1F00D2A2A0) = vpiNet
 vpi_get (vpiType, 00005F1F00D2A2A0) = vpiNet
 vpi_get (vpiType, 00005F1F00D2A2A0) = vpiNet
 vpi_get (vpiSize, 00005F1F00D2A2A0) = 32
 vpi_get (vpiVector, 00005F1F00D2A2A0) = 1
 vpi_iterate (vpiRange, 00005F1F00D2A2A0) = 0000000000000000
 vpi_handle (vpiLeftRange, 00005F1F00D2A2A0) = 00005F1F00D9DC40
 vpi_get_value (00005F1F00D9DC40, {format=Int}) = 31
 vpi_handle (vpiRightRange, 00005F1F00D2A2A0) = 00005F1F00D7C0E0
 vpi_get_value (00005F1F00D7C0E0, {format=Int}) = 0
 vpi_get_str (vpiType, 00005F1F00D2A2A0) = "???"
 vpi_register_cb ({reason=cbAfterDelay, obj=0000000000000000, time=50ns}) = 00005F1F00BED100
 vpi_register_cb ({reason=cbReadWriteSynch, obj=0000000000000000}) = 00005F1F00966DA0
vpi end callback 00005F1F006EF600
vpi call callback 00005F1F00966DA0 cbReadWriteSynch
 vpi_free_object (00005F1F00966DA0)
 vpi_get (vpiType, 00005F1F00C88710) = vpiNet
 vpi_put_value (00005F1F00C88710, 15)
 vpi_get (vpiType, 00005F1F00D2A2A0) = vpiNet
 vpi_put_value (00005F1F00D2A2A0, 1337)
vpi end callback 00005F1F00966DA0
vpi call callback 00005F1F00BED100 cbAfterDelay
 vpi_free_object (00005F1F00BED100)
 vpi_get_value (00005F1F00C88710, {format=BinStr}) = "11110000111100001111000011110000"
 vpi_get_value (00005F1F00D2A2A0, {format=BinStr}) = "00000000000000000000010100111001"
 vpi_get_value (00005F1F00D2A2A0, {format=BinStr}) = "00000000000000000000010100111001"
 vpi_get_value (00005F1F00C88710, {format=BinStr}) = "11110000111100001111000011110000"
 vpi_register_cb ({reason=cbAfterDelay, obj=0000000000000000, time=1fs}) = 00005F1F00D782C0
vpi end callback 00005F1F00BED100
vpi call callback 00005F1F00D782C0 cbAfterDelay
 vpi_free_object (00005F1F00D782C0)
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 50000001fs
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 50000001fs
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 50000001fs
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 50000001fs
 vpi_control (vpiFinish, 1)
vpi end callback 00005F1F00D782C0
vpi call callback 00005F1F00713750 cbEndOfSimulation
vpi end callback 00005F1F00713750
```

Relevant parts:
 ```
vpi_get (vpiType, 00005F1F00C88710) = vpiNet
 vpi_put_value (00005F1F00C88710, 15)
 vpi_get (vpiType, 00005F1F00D2A2A0) = vpiNet
 vpi_put_value (00005F1F00D2A2A0, 1337)
...
 vpi_get_value (00005F1F00D2A2A0, {format=BinStr}) = "00000000000000000000010100111001"
 vpi_get_value (00005F1F00C88710, {format=BinStr}) = "11110000111100001111000011110000"
```

I don't know if this is due to GHDL or cocotb not working properly.

ktbarrett, 2024-02-27:
I see the vpi_put_value on the signal and the later vpi_get_value returning an unexpected string. 99% sure this is a GHDL issue.

ktbarrett, 2026-01-09:
No report has been made. But basically it values are deposited on the handle, but GHDL seems to be losing it somewhere and ignoring the value set.

juliusgodo, 2026-01-09:
If I remember correctly, I made some very hacky work-arounds back then.

This change fixed it for my usecase: https://github.com/ghdl/ghdl/commit/d21fd6a33407a8e40b0fdf77ae9e95d3a1319a92