cocotb/cocotb #3746: CI: mingw fails to compile Icarus Verilog and hangs after "mkdir dep"

imphil, 2024-02-28:
Currently, Windows CI fails to compile Icarus Verilog; since the job times out:

```
Wed, 28 Feb 2024 18:57:22 GMT g++ -shared -Wl,--enable-auto-image-base  -o sizer.tgt sizer.o scan_lpms.o scan_logs.o -L.. -livl
Wed, 28 Feb 2024 18:57:22 GMT make[1]: Leaving directory '/d/a/cocotb/cocotb/iverilog/tgt-sizer'
Wed, 28 Feb 2024 18:57:22 GMT make[1]: Entering directory '/d/a/cocotb/cocotb/iverilog/driver'
Wed, 28 Feb 2024 18:57:22 GMT mkdir dep
Wed, 28 Feb 2024 19:29:43 GMT Error: The operation was canceled.
```

A diff between the [last passing](https://github.com/cocotb/cocotb/actions/runs/8031851937/job/21940743798) release run and the [first failing](https://github.com/cocotb/cocotb/actions/runs/8031854068/job/21940795642) release run is here:
[passing-failing.diff.txt](https://github.com/cocotb/cocotb/files/14440210/passing-failing.diff.txt)

Observations:
* In all cases I have seen the timeout happens after one of the "mkdir dep" steps.
* From the last passing run, the next step after the failing command should be ` gcc -I. -I..  -DHAVE_CONFIG_H -Wall -Wextra -Wshadow -Wstrict-prototypes -D__USE_MINGW_ANSI_STDIO=1  -g -O2 -MD -c -DIVL_ROOT='"/c/iverilog/lib/ivl"' -DIVL_SUFFIX='""' -DIVL_INC='"/c/iverilog/include"' -DIVL_LIB='"/c/iverilog/lib"' -DDLLIB='""' -DIVL_INCLUDE_INSTALL_DIR="\"\"" ./main.c`, which looks like a totally normal gcc call.
* There are no noticeable software updates between a passing and a failing run.
* The first failing command conveniently [updated only pre-commit hooks](https://github.com/cocotb/cocotb/commit/6511d5a9e1c49937583479e7ef3ff55465dc06f7), i.e., there's no actual cocotb code change between the last passing and the first failing CI run.
* The diff shows slightly different ordering of commands, but it looks more like stdout interleaving being different than actual differences in execution (it's a parallel make, so who knows).

I couldn't see anything actionable in the CI logs to look further. It might well be a host update on the runner that isn't reflected in the image version number displayed by the GH Runner.

imphil, 2024-02-28:
Since we need to get CI back green, I'll just disable the mingw tests (again).

imphil, 2024-02-28:
Oh wait, there was a passing test run 20 minutes ago at https://github.com/cocotb/cocotb/actions/runs/8087291013/job/22098910736?pr=3733. Maybe MSFT resolved the issue? Let's rerun some older runs ...

ktbarrett, 2024-02-28:
@imphil the failure seems random. It's been hit or miss, mostly miss, since we reintroduced it.