cocotb/cocotb #3747: VPI and friends' callback cleanup is messy

ktbarrett, 2024-02-28:
The mess of `gpi_cb_state_t`, `handle_xxx_callback`, `cleanup_callback` and `set_call_state` could potentially be massively simplified by removing the whole thing.

I think this was originally added to work around issues with registering and de-registering callbacks in the same callback cycle. This should be re-evaluated and not necessarily placed into the core logic of the callback objects.

ktbarrett, 2024-07-23:
So I can't guarantee this is the way it is, but it seems that this is the way it was intended to be.

Four states:
* GPI_FREE: no callback registered.
* GPI_PRIMED: callback registered and pending.
* GPI_CALL: callback is running.
* GPI_DELETE: callback deregistered but not cleaned up yet.

* GPI_FREE -> GPI_PRIMED: on `arm_callback`.
* GPI_PRIMED -> GPI_CALL: when the callback fires and `run_callback` is called.
* GPI_PRIMED -> GPI_DELETE: when the callback is diabled before it fires. It is also deregistered with the interface, but that isn't foolproof, so setting the state ensures that if the callback comes in, it is squashed. Occurs when `cleanup_callback` is called.
* GPI_CALL -> GPI_DELETE: when the callback is disabled after it fires. We can't cleanup the callback yet since we may re-prime it still. Occurs when `cleanup_callback` is called.
* GPI_CALL -> GPI_PRIMED: when the callback is enabled after it fires in `arm_callback`.
* GPI_DELETE -> GPI_FREE: after the callback run is complete and the callback is in a disabled state, so it can be deregistered with the interface and its memory cleaned up.

ktbarrett, 2024-07-24:
The obvious issue is that `cleanup_callback` is used for several functions and called from multiple locations. I think the easiest solution is to create a `disarm_callback`, which is intended to put the callback into the `GPI_DELETE` state regardless if it's primed or called. And that this function is only called from cocotb context, not from the simulator context. That leaves `cleanup_callback` to only be called by the simulator.

cocotb context functions:
* `arm_callback`
* `disarm_callback`

simulator context functions:
* `run_callback`
* cleanup_callback`

ktbarrett, 2024-07-24:
A follow on deletes `GPI_CALL`, because it's completely useless.

marlonjames, 2024-07-24:
I would also suggest refactoring `arm_callback` to use a switch statement with all of the states, for clarity.
I did some of that in #2349 but never extracted it to another PR.

ktbarrett, 2024-07-24:
This is what I got so far:
```c++
int GpiCommonCbHdl::run_callback() {
    switch (get_call_state()) {
        case GPI_PRIMED: {
            set_call_state(GPI_DELETE);
            this->gpi_function(m_cb_data);
            return 0;
        }
        case GPI_DELETE: {
            // callback was disarmed, but still fired, ignore
            return 0;
        }
        case GPI_FREE: {
            unreachable();
        }
    }
}

int GpiCommonCbHdl::arm_callback() {
    switch (get_call_state()) {
        case GPI_FREE: {
            // newly created, register with interface
            set_call_state(GPI_PRIMED);
            return 0;
        }
        case GPI_PRIMED: {
            // already primed, noop
            return 0;
        }
        case GPI_DELETE: {
            set_call_state(GPI_PRIMED);
            return 0;
        }
    }
}

int GpiCommonCbHdl::disarm_callback() {
    switch (get_call_state()) {
        case GPI_PRIMED: {
            set_call_state(GPI_DELETE);
            return 0;
        }
        case GPI_DELETE: {
            // already disarmed, noop
            return 0;
        }
        case GPI_FREE: {
            unreachable();
        }
    }
}

int GpiCommonCbHdl::cleanup_callback() {
    switch (get_call_state()) {
        case GPI_PRIMED: {
            // callback was rearmed, don't cleanup
            return 0;
        }
        case GPI_DELETE: {
            // deregister with interface
            delete this;  // I can't believe I finally got to write this
            return 0;
        }
        case GPI_FREE: {
            unreachable();
        }
    }
}
```

ktbarrett, 2024-07-24:
I'm actually thinking that `arm_callback` should return a handle and `disarm_callback` should take a handle. This is to support multiple callbacks. And of course `set_user_data` on the callback should be set by `arm_callback`. 

That parallels how Triggers, `_prime` and `_unprime`, and callbacks work (at least after my PR gets pushed up). We should re-envision the `CbHdl` objects not as the callback themselves, but like Triggers (Futures).