cocotb/cocotb #3754: Verilator extended identifiers

ktbarrett, 2024-03-04:
From testing #1610. In verilator, signals with extended identifiers are discovered with the `vpiName` starting with the backslash and ending with a space.

```python
dut._discover_all()
for name in dut._keys():
    print(repr(name))
```

Output:
```python
...
'\\(.*|this_looks_like_a_regex) '
'\\weird.signal[1] '
'\\weird.signal[2] '
...
```

This is an upstream problem.

marlonjames, 2024-03-05:
The standard doesn't give any guidance on using escaped identifiers with VPI.
It simply states states that the leading backslash and terminating white space are not considered part of the identifier.

For `vpiFullName`, because the hierarchy separator is a `.`, I would expect the escaped identifiers to include the backslash and whitespace, at least for the levels higher in the hierarchy. For `vpiName`, I think either could be correct.

iverilog appears to strip them for `vpiName`: https://github.com/steveicarus/iverilog/blob/master/ivtest/vpi_gold/br_gh317.gold

ktbarrett, 2024-03-05:
Looks like we should handle this somewhere between the GPI (ideal) and the Python handle objects (less than ideal). But we should normalize the names so they are easier to look up.

marlonjames, 2024-03-05:
It would be nice to handle hierarchical paths as an array/slice of strings rather than as a single string with possible escaped identifiers between some common separator. Especially because each GPI implementation has some different ways of handling hierarchy separators.

ktbarrett, 2024-03-05:
Should we make a 3.0 milestone?