cocotb/cocotb #3764: undeclared identifier 'basename' in verilator.cpp

rasgs, 2024-03-09:
* cocotb version: 2.0.0.dev0+318ec4b2
* OS: macOS Sonoma 14.2.1
* Simulator: Verilator 5.006 2023-01-22
* Python: Python 3.9.18

When running runner.build(...) I get the following error:
```
/Users/rgs/opt/miniforge3/envs/sdr/lib/python3.9/site-packages/cocotb/share/lib/verilator/verilator.cpp:77:21: error: use of undeclared identifier 'basename'
                    basename(argv[0]), traceFile);
```

I can solve it by including libgen.h in verilator.cpp:
`#include <libgen.h>`

ktbarrett, 2024-03-10:
I guess the question is whether to keep `basename` or remove it. This makes it so the use of verilator depends upon a POSIX environment. Does verilator already assume that?

ktbarrett, 2024-03-19:
The stable version of Verilator apparently doesn't build on Mac OS 11 ([see here](https://github.com/cocotb/cocotb/actions/runs/8335458211/job/22811065256?pr=3785)), even though the Verilator website says it supports it. I don't have a Mac laying around. I'm going to need some help from you to debug it.

rasgs, 2024-03-19:
Sure thing, I'd be happy to help! Any particulay way I should do it?

closed 2024-03-20 by ktbarrett

ktbarrett, 2024-03-20:
Apparently verilator master just doesn't support old versions of Mac OS. It was an easy fix.

PR #3785 (2024-03-20): Test Verilator on MacOS and fix it.
Closes #3764. Closes #3765.