cocotb/cocotb #3789: Replace `expect_fail` and `expect_error` with something like `pytest.mark.xfail`

ktbarrett, 2024-03-19:
This new decorator should be stackable, so we can specify multiple conditions when we expect the test to fail, what it will fail with (`Exception` or `assert`), and a reason string (much like #3709).

imphil, 2024-03-19:
I actually would love to see support for test waiver files. Right now, we code tool-version-specific expectations in the tests and then update them from time to time when a new tool release is available. It would be nice if we had a waiver/expectations file for, say, Questa 2023.4. When we add support for a different Questa version, we can start by copying that file, but then need to consciously update areas where we see deviations. More thought is needed, but it's something I had one the back of my head for a long time now.

cmarqu, 2024-03-20:
> it's something I had one the back of my head for a long time now.

At least since 2020 when you opened #1872 :)

ktbarrett, 2024-06-27:
@imphil That sounds better for us. I think we should still do this for something better for users.