cocotb/cocotb #3793: Cocotb equivalent to run_until_complete

timothyscherer, 2024-03-20:
I am currently trying to write a BFM for an existing Python API that doesn't use async/await. I have it working such that in one mode, I import the original API, then in another mode I import my implementation for cocotb. This part is working well. 

My only problem now is that, in order to match the API, none of the functions should be async. But I still need to call async functions from within.

I am not very familiar with Python async, but I believe [this](https://github.com/python/cpython/blob/main/Lib%2Fasyncio%2Fbase_events.py#L685) does what I think I need.

Also see [this](https://discuss.python.org/t/calling-coroutines-from-sync-code/23027) for reference (and my inspiration for trying to figure this out).

marlonjames, 2024-03-20:
Take a look at #3615 and #3650.

ktbarrett, 2024-03-20:
Is there a reason not to make an adapter around the existing Python BFM API to make them all `async`? That's going to be an easier route than the other way.