cocotb/cocotb #3795: Corrupted size vs Prev size. Aborted (core dumped) error

ttDuong282, 2024-03-21:
Hi everyone, 
I got stuck when I ran my Makefile for my design. My testbench Python code had nothing in it because I temporarily commented out all the calculations and memory production. However, when I ran "make", the terminal ran a lot of ccache g++ lines (I didn't understand how they worked), and the result was below. I thought that Cocotb seemed to have a size limit available for user designs, and mine exceeded it. How can I check the "Prev size" and "Corrupted size"? Thank you all for supporting me!
```ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable       -include Vtop__pch.h.fast -c -o Vtop___024root__DepSet_heccd7ead__10.o Vtop___024root__DepSet_heccd7ead__10.cpp
ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable       -include Vtop__pch.h.fast -c -o Vtop___024root__DepSet_heccd7ead__11.o Vtop___024root__DepSet_heccd7ead__11.cpp
ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable       -include Vtop__pch.h.fast -c -o Vtop__Dpi.o Vtop__Dpi.cpp
ccache g++   -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable       -x c++-header Vtop__pch.h -o Vtop__pch.h.slow.gch
ccache g++   -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable       -include Vtop__pch.h.slow -c -o Vtop___024root__Slow.o Vtop___024root__Slow.cpp
ccache g++   -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable       -include Vtop__pch.h.slow -c -o Vtop___024root__DepSet_h84412442__0__Slow.o Vtop___024root__DepSet_h84412442__0__Slow.cpp
ccache g++   -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable       -include Vtop__pch.h.slow -c -o Vtop___024root__DepSet_heccd7ead__0__Slow.o Vtop___024root__DepSet_heccd7ead__0__Slow.cpp
ccache g++   -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable       -include Vtop__pch.h.slow -c -o Vtop___024root__DepSet_heccd7ead__1__Slow.o Vtop___024root__DepSet_heccd7ead__1__Slow.cpp
ccache g++   -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable       -include Vtop__pch.h.slow -c -o Vtop__Syms.o Vtop__Syms.cpp
echo "" > Vtop__ALL.verilator_deplist.tmp
Archive ar -rcs Vtop__ALL.a Vtop.o Vtop___024root__DepSet_h84412442__0.o Vtop___024root__DepSet_heccd7ead__0.o Vtop___024root__DepSet_heccd7ead__1.o Vtop___024root__DepSet_heccd7ead__2.o Vtop___024root__DepSet_heccd7ead__3.o Vtop___024root__DepSet_heccd7ead__4.o Vtop___024root__DepSet_heccd7ead__5.o Vtop___024root__DepSet_heccd7ead__6.o Vtop___024root__DepSet_heccd7ead__7.o Vtop___024root__DepSet_heccd7ead__8.o Vtop___024root__DepSet_heccd7ead__9.o Vtop___024root__DepSet_heccd7ead__10.o Vtop___024root__DepSet_heccd7ead__11.o Vtop__Dpi.o Vtop___024root__Slow.o Vtop___024root__DepSet_h84412442__0__Slow.o Vtop___024root__DepSet_heccd7ead__0__Slow.o Vtop___024root__DepSet_heccd7ead__1__Slow.o Vtop__Syms.o
g++     verilator.o verilated.o verilated_dpi.o verilated_vpi.o verilated_threads.o Vtop__ALL.a   -Wl,-rpath,/home/ttduong/.local/lib/python3.8/site-packages/cocotb/libs -L/home/ttduong/.local/lib/python3.8/site-packages/cocotb/libs -lcocotbvpi_verilator  -pthread -lpthread -latomic   -o Vtop
rm Vtop__ALL.verilator_deplist.tmp
make[2]: Leaving directory '/workspace/cocotb/tmp/verilator_Tb/core_new/sim_build'
rm -f results.xml
MODULE=test_neuron_core_256x256 TESTCASE= TOPLEVEL=neuron_core_256x256 TOPLEVEL_LANG=verilog \
         sim_build/Vtop 
corrupted size vs. prev_size
make[1]: *** [/home/ttduong/.local/lib/python3.8/site-packages/cocotb/share/makefiles/simulators/Makefile.verilator:65: results.xml] Aborted (core dumped)
make[1]: Leaving directory '/workspace/cocotb/tmp/verilator_Tb/core_new'
make: *** [/home/ttduong/.local/lib/python3.8/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2
```
My makefile:
```
TOPLEVEL_LANG ?= verilog
SIM ?= verilator
TOPLEVEL := neuron_core_256x256

PWD=$(shell pwd)

COCOTB?=$(PWD)/../../..

MODULE = test_neuron_core_256x256

ifeq ($(TOPLEVEL_LANG),verilog)
    VERILOG_SOURCES = $(COCOTB)/tmp/verilator_Tb/core_new/hdl/neuron_core_256x256.v \
                      $(COCOTB)/tmp/verilator_Tb/core_new/hdl/address_decoder.v \
                      $(COCOTB)/tmp/verilator_Tb/core_new/hdl/choose_weight.v \
                      $(COCOTB)/tmp/verilator_Tb/core_new/hdl/core_decoder.v \
                      $(COCOTB)/tmp/verilator_Tb/core_new/hdl/DFFRAM256x32.v \
                      $(COCOTB)/tmp/verilator_Tb/core_new/hdl/neuron_block.v \
                      $(COCOTB)/tmp/verilator_Tb/core_new/hdl/neuron_slice.v \
                      $(COCOTB)/tmp/verilator_Tb/core_new/hdl/neuron_parameter.v \
                      $(COCOTB)/tmp/verilator_Tb/core_new/hdl/neuron_spike_out.v \
                      $(COCOTB)/tmp/verilator_Tb/core_new/hdl/slice_decoder.v \
                      $(COCOTB)/tmp/verilator_Tb/core_new/hdl/synapse_matrix.v 

    
else ifeq ($(TOPLEVEL_LANG),vhdl)
    VHDL_SOURCES =  $(COCOTB)/tests/designs/synapse_matrix/synapse_matrix_design.vhd

    ifneq ($(filter $(SIM),ius xcelium),)
        COMPILE_ARGS += -v93
    endif
else
    $(error A valid value (verilog or vhdl) was not provided for TOPLEVEL_LANG=$(TOPLEVEL_LANG))
endif

include $(shell cocotb-config --makefiles)/Makefile.sim
```

ktbarrett, 2024-03-21:
I think the easiest solution is to run `make clean` before running verilator again. This seems to be corruption due to linking stale objects.

ttDuong282, 2024-03-22:
@ktbarrett Thanks for your advice! However, I tried "make clean" before "make" but it still didn't work. The notice "corrupted size vs prev size" was still there!

ktbarrett, 2024-03-22:
Another possible cause is memory access issues like use after free, writing past the end of a malloc'd array clobbering the metadta of the next node, double frees... all to say that either cocotb or verilator are doing some bad stuff. I'm not sure which version of verilator you are using, but try updating it to the most recent release. If not, debugging with valgrind will be necessary.

ttDuong282, 2024-03-24:
@ktbarrett I haven't resolved this error since I read your response. My Verilator version is 5.019 rev v5.018-58-g7167d4bdc. I think I it's okay. But when considering debugging with valgrind, I found that there's nothings can help me solve this. I think the problem is the makefile which have the configured size because when I try running a slice of my design (it includes 8 slides), it really works. But I can't find any source describing about the supported size or something like that. :(