cocotb/cocotb #3801: Verilator fails test_dff.py example

dpetrisko, 2024-03-22:
Hello, the default example fails in Verilator (5.022, though it shouldn't matter).

```
# test_dff.py

import random

import cocotb
from cocotb.clock import Clock
from cocotb.triggers import RisingEdge
from cocotb.types import LogicArray

@cocotb.test()
async def dff_simple_test(dut):
    """Test that d propagates to q"""

    # Assert initial output is unknown
    assert LogicArray(dut.q.value) == LogicArray("X")
    # Set initial input value to prevent it from floating
    dut.d.value = 0

    clock = Clock(dut.clk, 10, units="us")  # Create a 10us period clock on port clk
    # Start the clock. Start it low to avoid issues on the first RisingEdge
    cocotb.start_soon(clock.start(start_high=False))

    # Synchronize with the clock. This will regisiter the initial `d` value
    await RisingEdge(dut.clk)
    expected_val = 0  # Matches initial input value
    for i in range(10):
        val = random.randint(0, 1)
        dut.d.value = val  # Assign the random value val to the input port d
        await RisingEdge(dut.clk)
        assert dut.q.value == expected_val, f"output q was incorrect on the {i}th cycle"
        expected_val = val # Save random value for next RisingEdge

    # Check the final input on the next clock
    await RisingEdge(dut.clk)
    assert dut.q.value == expected_val, "output q was incorrect on the last cycle"
```

The issue is that Verilator does not support X's, so the initial value is set by Verilator's 
```        --x-initial <mode>          Assign initial Xs to this value```
```        --x-assign <mode>           Assign non-initial Xs to this value```

Maybe it makes sense to set x-assign for verilator in Makefile.verilator and then modify the backend function of 'X' comparisons to use verilator's x-assign value?

ktbarrett, 2024-03-22:
This is already fixed on master.

dpetrisko, 2024-03-22:
Oh, I see that the example has been fixed. However, the README still has the outstanding bug.  Thanks!

ktbarrett, 2024-03-22:
Oh... totally missed that. Good catch!

ktbarrett, 2024-04-10:
I don't think we want simulator ifdefs in our examples. I will refactor the example and paste it on the README.

ktbarrett, 2025-08-27:
This was completed at some point...

closed 2025-08-27 by ktbarrett