cocotb/cocotb #3809: No module named 'cocotb_tools'

Jaegerseki, 2024-03-25:
rm -f results.xml
make -f Makefile results.xml
make[1]: Entering directory '/home/jalen/cocotb-master/cocotb-master/examples/adder/tests'
mkdir -p sim_build
/usr/bin/iverilog -o sim_build/sim.vvp -D COCOTB_SIM=1 -s adder -f sim_build/cmds.f -g2012   /home/jalen/cocotb-master/cocotb-master/examples/adder/tests/../hdl/adder.sv
rm -f results.xml
MODULE=test_adder TESTCASE= TOPLEVEL=adder TOPLEVEL_LANG=verilog \
         /usr/bin/vvp -M /home/jalen/.local/lib/python3.10/site-packages/cocotb/libs -m libcocotbvpi_icarus   sim_build/sim.vvp 
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:76   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on Icarus Verilog version 11.0 (stable)
     0.00ns INFO     cocotb                             Running tests with cocotb v1.8.1 from /home/jalen/.local/lib/python3.10/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1711367600
     0.00ns CRITICAL cocotb.regression                  Failed to import module test_adder: No module named 'cocotb_tools'
     0.00ns INFO     cocotb.regression                  MODULE variable was "test_adder"
     0.00ns INFO     cocotb.regression                  Traceback: 
     0.00ns INFO     cocotb.regression                  Traceback (most recent call last):
                                                          File "/home/jalen/.local/lib/python3.10/site-packages/cocotb/regression.py", line 203, in _discover_tests
                                                            module = _my_import(module_name)
                                                          File "/home/jalen/.local/lib/python3.10/site-packages/cocotb/regression.py", line 72, in _my_import
                                                            mod = __import__(name)
                                                          File "/home/jalen/.local/lib/python3.10/site-packages/_pytest/assertion/rewrite.py", line 178, in exec_module
                                                            exec(co, module.__dict__)
                                                          File "/home/jalen/cocotb-master/cocotb-master/examples/adder/tests/test_adder.py", line 11, in <module>
                                                            from cocotb_tools.runner import get_runner
                                                        ModuleNotFoundError: No module named 'cocotb_tools'
                                                        
Traceback (most recent call last):
  File "/home/jalen/.local/lib/python3.10/site-packages/cocotb/__init__.py", line 233, in _initialise_testbench
    _initialise_testbench_(argv_)
  File "/home/jalen/.local/lib/python3.10/site-packages/cocotb/__init__.py", line 316, in _initialise_testbench_
    regression_manager = RegressionManager.from_discovery(top)
  File "/home/jalen/.local/lib/python3.10/site-packages/cocotb/regression.py", line 174, in from_discovery
    return cls(dut, tests)
  File "/home/jalen/.local/lib/python3.10/site-packages/cocotb/regression.py", line 153, in __init__
    for test in tests:
  File "/home/jalen/.local/lib/python3.10/site-packages/cocotb/regression.py", line 203, in _discover_tests
    module = _my_import(module_name)
  File "/home/jalen/.local/lib/python3.10/site-packages/cocotb/regression.py", line 72, in _my_import
    mod = __import__(name)
  File "/home/jalen/.local/lib/python3.10/site-packages/_pytest/assertion/rewrite.py", line 178, in exec_module
    exec(co, module.__dict__)
  File "/home/jalen/cocotb-master/cocotb-master/examples/adder/tests/test_adder.py", line 11, in <module>
    from cocotb_tools.runner import get_runner
ModuleNotFoundError: No module named 'cocotb_tools'
Traceback (most recent call last):
  File "/home/jalen/.local/lib/python3.10/site-packages/cocotb/__init__.py", line 332, in _sim_event
    scheduler.log.error(msg)
AttributeError: 'NoneType' object has no attribute 'log'
     0.00ns ERROR    gpi                                Passing event to upper layer failed
VCD info: dumpfile dump.vcd opened for output.

marlonjames, 2024-03-25:
Checkout out the `v1.8.1` tag in the repository, that will give you the version of the example test that is compatible with the version of cocotb you are running.

Jaegerseki, 2024-03-25:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

marlonjames, 2024-03-27:
Reopen if you have further problems.

closed 2024-03-27 by marlonjames

luck-codeer, 2024-07-20:
v1.8.0 may be good