cocotb/cocotb #3814: GHDL Arrays Access

gbowers4, 2024-03-28:
# Environment

- GHDL         4.0.0 Built from GCC
- cocotb 1.8.1
- Ubuntu 22.04

# Issue

GHDL 4.0.0 Recently added support for unconstrained array typedefs in VHDL - Great! However as it seems this features hasn't been tested in cocotb a bug occurs. when a constrained array is read from/written to GHDL performs the same as any other simulator, however when the dimensions of the array type is unconstrained a cocotb GPI error occurs - I am concerned this is once again a GHDL VPI error

# Test

```python
@cocotbTest()
async def run(dut):
    """
        Template Test program
    """
    print(dut.test_constrained[0].value)
    print(dut.test_unconstrained[0].value)
```
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity rtl_bench is
port(
  clk : in std_logic
);
end rtl_bench;

architecture arch of rtl_bench is

  type constrained_array_t is array (0 to 3) of std_logic_vector(7 downto 0);
  type unconstrained_array_t is array (integer range <>) of std_logic_vector;
  signal test_constrained   : constrained_array_t := (others => x"ba");
  signal test_unconstrained : unconstrained_array_t (0 to 3)(7 downto 0) := (others => x"ba");

begin

end arch;
```

# Output

```
INFO     cocotb:simulator.py:305      0.00ns DEBUG    gpi                                VPI: Created GPI object from type ???(114)
INFO     cocotb:simulator.py:305      0.00ns DEBUG    gpi                                Checking rtl_bench.test_constrained exists
INFO     cocotb:simulator.py:305      0.00ns DEBUG    cocotb.test_constrained            Created
INFO     cocotb:simulator.py:305      0.00ns DEBUG    gpi                                Checking if index 0 native through implementation VPI
INFO     cocotb:simulator.py:305      0.00ns DEBUG    gpi                                VPI: Indexable object initialized with range [7:0] and length >8<
INFO     cocotb:simulator.py:305      0.00ns DEBUG    gpi                                VPI: test_constrained[0] initialized with 8 elements
INFO     cocotb:simulator.py:305      0.00ns DEBUG    gpi                                VPI: Created GPI object from type ???(36)
INFO     cocotb:simulator.py:305      0.00ns DEBUG    gpi                                Checking rtl_bench.test_constrained[0] exists
INFO     cocotb:simulator.py:305      0.00ns DEBUG    cocotb.test_constrained[0]         Created
INFO     cocotb:simulator.py:305 10111010
INFO     cocotb:simulator.py:305      0.00ns DEBUG    gpi                                Searching for test_unconstrained
INFO     cocotb:simulator.py:305      0.00ns DEBUG    gpi                                VPI: Created GPI object from type ???(114)
INFO     cocotb:simulator.py:305      0.00ns DEBUG    gpi                                Checking rtl_bench.test_unconstrained exists
INFO     cocotb:simulator.py:305      0.00ns DEBUG    cocotb.test_unconstrained          Created
INFO     cocotb:simulator.py:305      0.00ns DEBUG    gpi                                Checking if index 0 native through implementation VPI
INFO     cocotb:simulator.py:305      0.00ns DEBUG    gpi                                Unable to find handle through vpi_handle_by_index(), attempting second method
INFO     cocotb:simulator.py:305      0.00ns DEBUG    gpi                                Unable to vpi_get_handle_by_index test_unconstrained[0]
INFO     cocotb:simulator.py:305      0.00ns WARNING  gpi                                Failed to find a handle at index 0 via any registered implementation
INFO     cocotb:simulator.py:305      0.00ns TRACE    gpi                                Returning to simulator
INFO     cocotb:simulator.py:305      0.00ns TRACE    gpi                                Returning control to simulator
INFO     cocotb:simulator.py:305      0.00ns TRACE    gpi                                Passing control to GPI user
INFO     cocotb:simulator.py:305      0.00ns TRACE    gpi                                Returning to Python
INFO     cocotb:simulator.py:305      0.00ns INFO     cocotb.regression                  run failed
INFO     cocotb:simulator.py:305                                                         Traceback (most recent call last):
INFO     cocotb:simulator.py:305                                                           File "/data/firmware/src/pylib/cocotb/templates/avl_st_template/avl_st_template_test.py", line 43, in run
INFO     cocotb:simulator.py:305                                                             print(dut.test_unconstrained[0].value)
INFO     cocotb:simulator.py:305                                                           File "/usr/local/lib/python3.10/dist-packages/cocotb/handle.py", line 652, in __getitem__
INFO     cocotb:simulator.py:305                                                             raise IndexError(
INFO     cocotb:simulator.py:305                                                         IndexError: test_unconstrained(GPI_ARRAY) contains no object at index 0
```

marlonjames, 2024-03-28:
Yes, when I looked at this not too long ago, there is missing support in GHDL's internal AVHPI for unconstrained arrays.
It doesn't appear too difficult to add, but it involves finding the Run Time Information (RTI) for the constraints that are associated with the signal.
I didn't have time or familiarity to figure that out, but if you want to dive into GHDL's internals, feel free.

gbowers4, 2024-03-28:
I will start to get myself familiarised, the world of the GHDL VPI is a bit alien to me but would be a great blocker to get rid of.