cocotb/cocotb #3815: vcs U-2023.03-SP1-1 plus conda python wont load

emunoz-at-tt, 2024-03-28:
Hey guys.

We are using cocotb 1.8 and also tried 1.8.1 on previous versions of vcs [ up to T versions ] we can load our environment and use cocotb. we are having a blast. 

However when we try versions on the U range we get this weird error , it wont even load 
`Fatal Python error: Py_Initialize: Unable to get the locale encoding
ModuleNotFoundError: No module named 'encodings
`

Googling around seems like a fundamental python not being properly found. i tried adding PYTHONHOME , PYtHONPATH, etc with the paths where python is installed in the conda area to no avail

Is there some way to see what is going on here?

This works perfectly with no warnings or errors on VCS T versions

emunoz-at-tt, 2024-03-28:
vcs command line:

`
/tools_vendor/synopsys/vcs//U-2023.03-SP1-1/linux64/bin/vcs1 -Mcc=gcc -Mcplusplus=g++ \                                                                                                                                                                                         │ 
-Masflags= -Mcfl= -pipe -fPIC -O -I/tools_vendor/synopsys/vcs//U-2023.03-SP1-1/include \
-Mxllcflags= -Mxcflags= -pipe -fPIC -I/tools_vendor/synopsys/vcs//U-2023.03-SP1-1/include \
-Mldflags= -Lvendor/yaml-cpp/build -rdynamic \
-Mout=/vcs-core/out/simv \
-Mamsrun="" -Mvcsaceobjs="" -Mobjects=" /tools_vendor/synopsys/vcs//U-2023.03-SP1-1/linux64/lib/libvirsim.so \
/tools_vendor/synopsys/vcs//U-2023.03-SP1-1/linux64/lib/liberrorinf.so /tools_vendor/synopsys/vcs//U-2023.03-SP1-1/linux64/lib/libsnpsmalloc.so \
/tools_vendor/synopsys/vcs//U-2023.03-SP1-1/linux64/lib/libvfs.so /proj_tensix/user_dev/emunoz/newrepo6/cmake-build-t6-trin-n1-vcs-cctb/src/t6ifc/vcs-core/tvm_tb/libvcs-core-tvm-tb.so \
-lboost_system -lyaml-cpp -lsqlite3 -lz -lglog -lstdc++fs " -Mexternalobj= -Msaverestoreobj=/tools_vendor/synopsys/vcs//U-2023.03-SP1-1/linux64/lib/vcs_save_restore_new.o \
-Mcrt0= -Mcrtn="" -Mcsrc="" -Mdir=/vcs-core/out/csrc \
-Msyslibs=/tools_vendor/synopsys/verdi//U-2023.03-SP1-1/share/PLI/VCS/LINUX64/pli.a \
-ldl -lm  -l vcs_compile.log -V -kdb -Xufe=2steps -lca +error+20 +vcs+lic+wait +define+ECC_ENABLE \
+define+ECC_ENABLE +define+MAILBOX_TARGET=6 -xprop=tmerge -full64 -timescale=1ns/1ps \
-error=PCWM-L +warn=noTFIPC +warn=noLCA_FEATURES_ENABLED -assert disable_cover -assert \
svaext -o /vcs-core/out/simv \
-P /vcs-core/cocotb_pli.tab -load \
/conda/conda-envs/tek/lib/python3.10/site-packages/cocotb/libs/libcocotbvpi_vcs.so \
+define+T_CCTB=1 +define+T_CCTB_1 -debug_acc+dmptf +vpi +vcsd1 +itf+/tools_vendor/synopsys/vcs//U-2023.03-SP1-1/linux64/lib/vcsdp_lite.tab \
-debug_region+cell+encrypt -debug_access +lint=all +warn=all -error=IWU -picarchive \
-P /tools_vendor/synopsys/verdi//U-2023.03-SP1-1/share/PLI/VCS/LINUX64/verdi.tab \
-fsdb -sverilog +vpi -gen_obj -f vcs.f

`

marlonjames, 2024-03-28:
In your Python environment, what does running `cocotb-config --libpython` give you?

emunoz-at-tt, 2024-03-28:
/conda/conda-envs/tek/lib/libpython3.10.so.1.0

marlonjames, 2024-03-28:
Are you setting `LIBPYTHON_LOC` environment variable to that path?

emunoz-at-tt, 2024-03-28:
yep:
`echo $LIBPYTHON_LOC
/conda/conda-envs/tek/lib/libpython3.10.so.1.0`

marlonjames, 2024-03-28:
Is cocotb installed from pypi or did you build it locally?

emunoz-at-tt, 2024-03-28:
From using conda. in the conda env file we have this 
`
name: tek
channels:
  - conda-forge
dependencies:
  - numpy
  - python=3.10.8
  - pyqt=5.12.3
  - sphinx
  - pip
  - numpy
  - python-gitlab
  - pymongo
  - pip:
    - watchdog
    - python-constraint
    - PyYAML
    - elasticsearch>=7,<8
    - cocotb
    - cocotb_coverage
    - pyvsc==0.8.7.6990678536
    - pytest
    - pyuvm
    - systemrdl-compiler
    - /proj_tensix/user_dev/pymodules/pysqlite3
`

marlonjames, 2024-03-28:
At this point, I'll point you to https://docs.python.org/3.10/c-api/intro.html#embedding-python which has some details on how the embedded Python interpreter determines search paths.

We may look at using [`Py_InitializeFromConfig`](https://docs.python.org/3/c-api/init_config.html#c.Py_InitializeFromConfig) for Python 3.8+.